commaai/openpilot #1: Lincoln MKZ / Ford Fusion compatibility

ingenieroariel, 2016-11-30:
Udacity is also building a self driving car using that model as a reference.

What is the list of things to take into account to port openpilot to other cars?

Here is a list based on what I have read so far:

 - [ ] Steering
 - [ ] Brake/Throttle 
 - [ ] Shift

ingenieroariel, 2016-11-30:
First thing seems to be to create a file like this one:

[image]

ingenieroariel, 2016-11-30:
Found one here: https://bitbucket.org/DataspeedInc/dbw_mkz_ros/downloads

jon-mullen, 2016-11-30:
That dbc's not correct for production cars, just (i assume) dataspeed's

geohot, 2016-11-30:
Unfortunately yes.

ingenieroariel, 2016-11-30:
I have a Ford Fusion 2017 myself and will debug further next week when my oneplus three arrives.

galv, 2016-11-30:
That dbc file won't work at all. It defines the message formats on a private CAN bus run by dataspeed. Dataspeed, as far as I know, controls the vehicle via some electromechanical methods, not just CAN codes. A node on the private CAN bus translates these messages to actual electrical signals. My guess is that the throttle is handled by controlling a voltage, but I'm not sure.

ingenieroariel, 2016-12-14:
According to the following spreadsheet [1] available from the OpenXC website here is the information we can find on CAN1 [OBD-II pins `6 (+) and 14 (-)` ]:  `CAN high / CAN low from ISO 15765-4 and SAE-J2284` 

steering_wheel_angle
engine_speed
transmission_gear_position
~gear_lever_position~
ignition_status
brake_pedal_status
~parking_brake_status~
headlamp_status
accelerator_pedal_position
torque_at_transmission
vehicle_speed
fuel_consumed_since_restart
door_status
windshield_wiper_status
odometer
high_beam_status
fuel_level
latitude
longitude
~button_event~

[1] https://docs.google.com/spreadsheets/d/1hOBi9-tFwR1KRFXfeaHTAddwJuSGx5Ir1ET4N2zWAiE/edit#gid=6

ingenieroariel, 2016-12-14:
For people trying to help solve this issue @geohot recommended the following document:

http://www.ioactive.com/pdfs/IOActive_Adventures_in_Automotive_Networks_and_Control_Units.pdf

I personally recommend to get an OpenXC adapter from Ford instead of trying to build the NEO can adapter. I got the FORD REFERENCE VI for $130 usd at https://shop.openxcplatform.com

ingenieroariel, 2016-12-17:
okay, I've been able to read data from the car via bluetooth and USB. So far I have only been able to get data in JSON format, but I plan to see if I can create a raw CAN dump using the kvaser leaf light.

Once I have a raw dump with the corresponding JSON dump I'll share it here.

[image]

```{"value":0.0,"name":"accelerator_pedal_position","a":{"a":{"name":"accelerator_pedal_position"},"b":-951985080},"timestamp":1481910779.317}
{"value":750.0,"name":"engine_speed","a":{"a":{"name":"engine_speed"},"b":1191599776},"timestamp":1481910779.319}
{"value":-253.5,"name":"steering_wheel_angle","a":{"a":{"name":"steering_wheel_angle"},"b":-1273841975},"timestamp":1481910779.323}
{"value":102.71772,"name":"fuel_level","a":{"a":{"name":"fuel_level"},"b":227280207},"timestamp":1481910779.323}
{"value":0.027625,"name":"fuel_consumed_since_restart","a":{"a":{"name":"fuel_consumed_since_restart"},"b":989500470},"timestamp":1481910779.359}
{"value":551.0,"name":"odometer","a":{"a":{"name":"odometer"},"b":2014857251},"timestamp":1481910779.368}
```
https://gist.github.com/ingenieroariel/ff1f4cac30e1bc5ec22f72967b1d2866

ingenieroariel, 2016-12-25:
Here is a sample data dump from a regular ford fusion when idle. It'll be useful later when comparing with the data dump from can bus 1 after cruise control is enabled.

```
{"bus": 1, "data": "0x0000000000000000", "id": 65, "timestamp": 1482689993.885337}                                    
{"bus": 1, "data": "0x140401f981f70000", "id": 390, "timestamp": 1482689993.885957}
{"bus": 1, "data": "0x8c00000006500000", "id": 1068, "timestamp": 1482689993.886489}
{"bus": 1, "data": "0xa000000000000000", "id": 1109, "timestamp": 1482689993.887}
{"bus": 1, "data": "0x7ef47ef4a174f000", "id": 145, "timestamp": 1482689993.887517}
{"bus": 1, "data": "0x7d0854008c804000", "id": 130, "timestamp": 1482689993.888006}
{"bus": 1, "data": "0x7d01800023b07cd7", "id": 133, "timestamp": 1482689993.888512}
{"bus": 1, "data": "0x2000000000000000", "id": 71, "timestamp": 1482689993.888998}
{"bus": 1, "data": "0x10c0000000000000", "id": 357, "timestamp": 1482689993.889519}
{"bus": 1, "data": "0x140c000000000000", "id": 369, "timestamp": 1482689993.890035}
{"bus": 1, "data": "0x00007fe17fe10000", "id": 512, "timestamp": 1482689993.890527}
{"bus": 1, "data": "0x04f9180060000000", "id": 514, "timestamp": 1482689993.891034}
{"bus": 1, "data": "0xc0007d01b3000000", "id": 516, "timestamp": 1482689993.891495}
{"bus": 1, "data": "0x0000000097000000", "id": 560, "timestamp": 1482689993.891988}
{"bus": 1, "data": "0x7d04002e660701b4", "id": 1071, "timestamp": 1482689993.892702}
{"bus": 1, "data": "0x0100273a00000000", "id": 936, "timestamp": 1482689993.893345}
{"bus": 1, "data": "0x002220ff0ff0eeee", "id": 938, "timestamp": 1482689993.894054}
{"bus": 1, "data": "0x125f0ff078eee000", "id": 939, "timestamp": 1482689993.894706}
{"bus": 1, "data": "0x0000000000000000", "id": 66, "timestamp": 1482689993.895373}
{"bus": 1, "data": "0x7d01800022c07cd7", "id": 133, "timestamp": 1482689993.897228}
{"bus": 1, "data": "0x0000000000400000", "id": 818, "timestamp": 1482689993.898004}
{"bus": 1, "data": "0x727fff100019fb00", "id": 359, "timestamp": 1482689993.898527}
{"bus": 1, "data": "0x04f8200060000000", "id": 514, "timestamp": 1482689993.898945}
{"bus": 1, "data": "0xc0007d01b3000000", "id": 516, "timestamp": 1482689993.899377}
{"bus": 1, "data": "0x0000fb40009fe7ff", "id": 125, "timestamp": 1482689993.90242}
{"bus": 1, "data": "0x000000000000fb40", "id": 532, "timestamp": 1482689993.90312}
{"bus": 1, "data": "0x0000e0f40ffe0ffe", "id": 1045, "timestamp": 1482689993.903668}
{"bus": 1, "data": "0x01008000400000fb", "id": 1200, "timestamp": 1482689993.904254}
{"bus": 1, "data": "0x0200000000000000", "id": 819, "timestamp": 1482689993.904832}
{"bus": 1, "data": "0x0000000000000000", "id": 65, "timestamp": 1482689993.905323}
{"bus": 1, "data": "0x005021d7cf000001", "id": 578, "timestamp": 1482689993.905862}
{"bus": 1, "data": "0x140401f981f70000", "id": 390, "timestamp": 1482689993.906278}
```

ingenieroariel, 2016-12-25:
Here is a full can dump with park assist activated:

```
   0x742000001dd20000 Bus 1: 0x166               Msgs: 735  Freq. (Hz): 2
    0x72805910001a0600 Bus 1: 0x167               Msgs: 342  Freq. (Hz): 10
    0x14803a10001a0800 Bus 1: 0x171               Msgs: 665              2
    0x00802810001a0400 Bus 1: 0x168               Msgs: 756  Freq. (Hz): 16
    0x0086c00000000000 Bus 1: 0x169               Msgs: 9    Freq. (Hz): 1
    0x002702000e95ce94 Bus 1: 0x178               Msgs: 525  Freq. (Hz): 2
    0x142402000e930000 Bus 1: 0x186               Msgs: 245  Freq. (Hz): 6
    0x1a0801fd0e920943 Bus 1: 0x179               Msgs: 72   Freq. (Hz): 2
    0x0a0640f981f700fc Bus 1: 0x202               Msgs: 572  Freq. (Hz): 3
    0x000080f981f70000 Bus 1: 0x200               Msgs: 374  Freq. (Hz): 11
    0xfcf4802f802fff41 Bus 1: 0x213               Msgs: 292  Freq. (Hz): 6
    0xccfa100175000000 Bus 1: 0x204               Msgs: 703  Freq. (Hz): 17
    0x000000007b00f4b0 Bus 1: 0x214               Msgs: 155  Freq. (Hz): 4
    0x0000000082000960 Bus 1: 0x216               Msgs: 454  Freq. (Hz): 14
    0x0000000000000c30 Bus 1: 0x217               Msgs: 455  Freq. (Hz): 14
    0xc1014b3931485232 Bus 1: 0x40a               Msgs: 436  Freq. (Hz): 13
    0x1006000097000000 Bus 1: 0x230               Msgs: 785  Freq. (Hz): 19
    0xe0022046d8000001 Bus 1: 0x242               Msgs: 96   Freq. (Hz): 2
    0x00002034d6000000 Bus 1: 0x261               Msgs: 180  Freq. (Hz): 4
    0x0000000000000000 Bus 1: 0x262               Msgs: 53   Freq. (Hz): 1
    0x0000510000000100 Bus 1: 0x2f1               Msgs: 46   Freq. (Hz): 1
    0x81a025b981c400c8 Bus 1: 0x326               Msgs: 75   Freq. (Hz): 1
    0xc01d5629047c9422 Bus 1: 0x331               Msgs: 73   Freq. (Hz): 2
    0x0000000900000000 Bus 1: 0x332               Msgs: 15   Freq. (Hz): 1
    0x0000000000000000 Bus 1: 0x333               Msgs: 61   Freq. (Hz): 1
    0x03fe00ffffff0400 Bus 1: 0x336               Msgs: 78   Freq. (Hz): 2
    0x00c0006aa7343000 Bus 1: 0x35e               Msgs: 12   Freq. (Hz): 1
    0x0000000906000000 Bus 1: 0x37b               Msgs: 90   Freq. (Hz): 1
    0x0000000000000000 Bus 1: 0x386               Msgs: 68   Freq. (Hz): 1
    0x2000000000000000 Bus 1: 0x38d               Msgs: 7    Freq. (Hz): 1
    0x1a00b80b00000000 Bus 1: 0x3a8               Msgs: 865  Freq. (Hz): 3
    0x0a00afdc0ff00000 Bus 1: 0x3aa               Msgs: 327  Freq. (Hz): 9
    0x12ff0ff040000000 Bus 1: 0x3ab               Msgs: 31   Freq. (Hz): 1
    0x4488c20c10000002 Bus 1: 0x3b3               Msgs: 24   Freq. (Hz): 1
    0x401111ff25250000 Bus 1: 0x3b4               Msgs: 16
    0x011c0121011b0112 Bus 1: 0x3b5               Msgs: 5    Freq. (Hz): 1
    0x08000b0000000000 Bus 1: 0x3b6               Msgs: 17   Freq. (Hz): 1
    0x0000090000000000 Bus 1: 0x3b7               Msgs: 28   Freq. (Hz): 1
    0xfffff3ff00005f8c Bus 1: 0x3b8               Msgs: 69   Freq. (Hz): 1
    0x5cc3000000000000 Bus 1: 0x3c3               Msgs: 14   Freq. (Hz): 1
    0x0003080400000000 Bus 1: 0x3c7               Msgs: 7    Freq. (Hz): 1
    0x0164000000000000 Bus 1: 0x3da               Msgs: 10   Freq. (Hz): 1
    0x0000000080000000 Bus 1: 0x3e0               Msgs: 5    Freq. (Hz): 1
    0x8000010701800000 Bus 1: 0x3e1               Msgs: 8    Freq. (Hz): 1
    0x0000000080000000 Bus 1: 0x3e2               Msgs: 6
    0x0001041380016400 Bus 1: 0x3e3               Msgs: 4    Freq. (Hz): 1
    0xc20414080f200000 Bus 1: 0x3eb               Msgs: 50   Freq. (Hz): 1
    0x0000000000006000 Bus 1: 0x3ea               Msgs: 3    Freq. (Hz): 1
    0x01023130163639ff Bus 1: 0x415               Msgs: 24   Freq. (Hz): 16
    0x0104014107f670c1 Bus 1: 0x41                Msgs: 47   Freq. (Hz): 1
    0x8000e0f400000000 Bus 1: 0x41d               Msgs: 106              3
    0x9a00d4f701000000 Bus 1: 0x416               Msgs: 359  Freq. (Hz): 9
    0x8100fc0006600000 Bus 1: 0x42c               Msgs: 652  Freq. (Hz): 2
    0x0000000000000000 Bus 1: 0x42                Msgs: 342  Freq. (Hz): 9
    0x0200900000009300 Bus 1: 0x421               Msgs: 74   Freq. (Hz): 1
    0x2000000000000700 Bus 1: 0x47                Msgs: 43   Freq. (Hz): 2
    0x00031f000000f000 Bus 1: 0x42d               Msgs: 144  Freq. (Hz): 3
    0x7d04002e66270176 Bus 1: 0x42f               Msgs: 346  Freq. (Hz): 2
    0x83000303861f0176 Bus 1: 0x430               Msgs: 214  Freq. (Hz): 5
    0x23e374000fff0203 Bus 1: 0x434               Msgs: 65   Freq. (Hz): 1
    0x28231097e1000103 Bus 1: 0x435               Msgs: 56   Freq. (Hz): 1
    0x2820189f00000000 Bus 1: 0x439               Msgs: 55   Freq. (Hz): 1
    0x3218506a38dd7414 Bus 1: 0x43c               Msgs: 6    Freq. (Hz): 1
    0x004b769000000000 Bus 1: 0x43d               Msgs: 68   Freq. (Hz): 2
    0x0046738d00000000 Bus 1: 0x43e               Msgs: 103  Freq. (Hz): 2
    0x2273000004000000 Bus 1: 0x447               Msgs: 104  Freq. (Hz): 2
    0xa000000000000000 Bus 1: 0x455               Msgs: 74   Freq. (Hz): 1
[... 6 lines cut ...]
    0x00c0d8d8d8d8c000 Bus 1: 0x4b                Msgs: 69   Freq. (Hz): 1
    0x0000004540144071 Bus 1: 0x5a                Msgs: 650  Freq. (Hz): 3
    0x20afffc030143074 Bus 1: 0x4c                Msgs: 318  Freq. (Hz): 11
    0x8100ffffffffffff Bus 1: 0x581               Msgs: 64   Freq. (Hz): 1
    0x9500ffffffffffff Bus 1: 0x595               Msgs: 9
[... cut ...]
    0x7ccb800075266dd6 Bus 1: 0x91                Msgs: 105              5
      74bb800083906dd7                                  744  Freq. (Hz): 24
    0x6ef87f017f8456c5 Bus 1: 0x92                Msgs: 290  Freq. (Hz): 7
        9f6f99736f9f7f   
```

If anyone wants to help me out identify what is what, be my guest. My next target is finding a can bus directly on the front camera.

PS: Not sure what the right etiquette for this project is, if I am spamming too much please let me know and I'll do less frequent updates.

adam-nelson, 2016-12-28:
You reckon this the same system for all ford vehicles.

I could try this in a focus

ingenieroariel, 2017-01-01:
[image]
I have connected the openxc device to the car and to the oneplus 3 via usb OTG. Next is looking into boardd and trying to read the JSON encoded raw can output coming from the openxc adapter.

ingenieroariel, 2017-01-19:
Okay, after cutting the wires from the canbus that went to the camera I was able to write a few packets that changed the left arrow keys on my wheel control. 

When connecting the camera again I got the following id codes that are potentially related to LKAS and collision assist.

`3ca`
`3cd`
`3d8`

I'll collect data from just those 3 codes on a longer ride today. Is anyone interested in helping me decode them?

Here is a sample in SocketCan format:

```
(1484854733.03) can1 3ca#fe7e458200000000
(1484854733.04) can1 3cd#0000000000000000
(1484854733.04) can1 3d8#0000000080456030
(1484854733.06) can1 3ca#fe7e458200000000
(1484854733.08) can1 3ca#fe7e458200000000
(1484854733.12) can1 3ca#fe7e458200000000
(1484854733.14) can1 3ca#fe7e458200000000
(1484854733.18) can1 3ca#fe7e458200000000
(1484854733.2) can1 3ca#fe7e458200000000
(1484854733.24) can1 3ca#fe7e458200000000
(1484854733.26) can1 3ca#fe7e458200000000
(1484854733.3) can1 3ca#fe7e458200000000
(1484854733.32) can1 3ca#fe7e458200000000
(1484854733.36) can1 3ca#fe7e458200000000
(1484854733.38) can1 3ca#fe7e458200000000
(1484854733.42) can1 3ca#fe7e458200000000
(1484854733.44) can1 3ca#fe7e458200000000
(1484854733.48) can1 3ca#fe7e458200000000
(1484854733.5) can1 3ca#fe7e458200000000
(1484854733.54) can1 3ca#fe7e458200000000
(1484854733.56) can1 3ca#fe7e458200000000
(1484854733.6) can1 3ca#fe7e458200000000
(1484854733.62) can1 3ca#fe7e458200000000
(1484854733.66) can1 3ca#fe7e458200000000
(1484854733.68) can1 3ca#fe7e458200000000
(1484854733.72) can1 3ca#fe7e458200000000
(1484854733.74) can1 3ca#fe7e458200000000
(1484854733.78) can1 3ca#fe7e458200000000
(1484854733.8) can1 3ca#fe7e458200000000
(1484854733.84) can1 3ca#fe7e458200000000
(1484854733.86) can1 3ca#fe7e458200000000
(1484854733.9) can1 3ca#fe7e458200000000
(1484854733.93) can1 3ca#fe7e458200000000
(1484854733.96) can1 3ca#fe7e458200000000
(1484854733.98) can1 3ca#fe7e458200000000
(1484854734.02) can1 3ca#fe7e458200000000
(1484854734.04) can1 3cd#0000000000000000
(1484854734.04) can1 3d8#0000000080456030
(1484854734.05) can1 3ca#fe7e458200000000
(1484854734.08) can1 3ca#fe7e458200000000
(1484854734.1) can1 3ca#fe7e458200000000
(1484854734.14) can1 3ca#fe7e458200000000
(1484854734.16) can1 3ca#fe7e458200000000
(1484854734.2) can1 3ca#fe7e458200000000
(1484854734.23) can1 3ca#fe7e458200000000
(1484854734.26) can1 3ca#fe7e458200000000
(1484854734.28) can1 3ca#fe7e458200000000
(1484854734.31) can1 3ca#fe7e458200000000
(1484854734.35) can1 3ca#fe7e458200000000
(1484854734.38) can1 3ca#fe7e458200000000
(1484854734.41) can1 3ca#fe7e458200000000
(1484854734.44) can1 3ca#fe7e458200000000
(1484854734.47) can1 3ca#fe7e458200000000
(1484854734.5) can1 3ca#fe7e458200000000
(1484854734.52) can1 3ca#fe7e458200000000
(1484854734.56) can1 3ca#fe7e458200000000
(1484854734.58) can1 3ca#fe7e458200000000
(1484854734.62) can1 3ca#fe7e458200000000
(1484854734.64) can1 3ca#fe7e458200000000
(1484854734.68) can1 3ca#fe7e458200000000
(1484854734.7) can1 3ca#fe7e458200000000
(1484854734.74) can1 3ca#fe7e458200000000
(1484854734.76) can1 3ca#fe7e458200000000
(1484854734.8) can1 3ca#fe7e458200000000
(1484854734.82) can1 3ca#fe7e458200000000
(1484854734.86) can1 3ca#fe7e458200000000
(1484854734.88) can1 3ca#fe7e458200000000
(1484854734.92) can1 3ca#fe7e458200000000
(1484854734.94) can1 3ca#fe7e458200000000
(1484854734.98) can1 3ca#fe7e458200000000
(1484854735.0) can1 3ca#fe7e458200000000
(1484854735.04) can1 3ca#fe7e458200000000
(1484854735.04) can1 3cd#0000000000000000
(1484854735.04) can1 3d8#0000000080456030
(1484854735.06) can1 3ca#fe7e458200000000
(1484854735.1) can1 3ca#fe7e458200000000
(1484854735.12) can1 3ca#fe7e458200000000
(1484854735.16) can1 3ca#fe7e458200000000
(1484854735.18) can1 3ca#fe7e458200000000
(1484854735.22) can1 3ca#fe7e458200000000
```

ingenieroariel, 2017-01-19:
BTW, here is a list of all the packet ids that are NOT coming from the front camera:

```background_ids = ['047', '04a', '04b', '04c', '05a', '05c', '05d', '076', '077', '078', '07d', '081', '082', '083', '084', '085', '091', '092', '165', '167', '168', '169', '178', '186', '187', '188', '18a', '200', '202', '204', '213', '214', '216', '217', '230', '242', '261', '265', '2a1', '33b', '350', '3a6', '3a7', '3a8', '3b3', '3c3', '3cc', '3e0', '3e2', '3e5', '3e6', '3eb', '40a', '415', '416', '41d', '41e', '422', '423', '42c', '42f', '430', '431', '43a', '453', '454', '455', '456', '4b0', '593', '596', '59e', '5b3', '5b5']```

ingenieroariel, 2017-01-20:
I may be wrong but here is what I think based on enabling LKAS on the road.

This message may mean LKAS is enabled but no lane has been found:

`3d8#0000000080456030`

In this frame LKAS found me drifting away from my lane and trying to get me back on track.

```
(1484874055.46) can0 3d8#0000000080456030
(1484874056.46) can0 3d8#0000000080456030
(1484874057.46) can0 3d8#0000000080456030
(1484874058.26) can0 3d8#0000000080856030
(1484874058.46) can0 3d8#0000000080856030
(1484874059.46) can0 3d8#0000000080856030
(1484874060.46) can0 3d8#0000000080856030
(1484874061.46) can0 3d8#0000000080856030
(1484874061.64) can0 3d8#0000000080855830
(1484874062.46) can0 3d8#0000000080855830
(1484874062.49) can0 3d8#0000000080856030
(1484874063.46) can0 3d8#0000000080856030
(1484874063.64) can0 3d8#0000000080855830
(1484874064.46) can0 3d8#0000000080855830
(1484874065.46) can0 3d8#0000000080855830
(1484874065.76) can0 3d8#0000000080853030
(1484874066.46) can0 3d8#0000000080853030
(1484874067.46) can0 3d8#0000000080853030
(1484874068.46) can0 3d8#0000000080853030
(1484874069.46) can0 3d8#0000000080853030
(1484874070.46) can0 3d8#0000000080853030
(1484874071.36) can0 3d8#0000000080c53030
(1484874071.46) can0 3d8#0000000080c53030
(1484874072.46) can0 3d8#0000000080c53030
(1484874073.46) can0 3d8#0000000080c53030
(1484874074.46) can0 3d8#0000000080c53030
(1484874075.46) can0 3d8#0000000080c53030
(1484874076.46) can0 3d8#0000000080c53030
(1484874077.46) can0 3d8#0000000080c53030
(1484874078.46) can0 3d8#0000000080c53030
(1484874079.46) can0 3d8#0000000080c53030
(1484874080.46) can0 3d8#0000000080c53030
(1484874081.46) can0 3d8#0000000080c53030
(1484874082.46) can0 3d8#0000000080c53030
(1484874083.46) can0 3d8#0000000080c53030
(1484874084.46) can0 3d8#0000000080c53030
(1484874085.46) can0 3d8#0000000080c53030
(1484874086.46) can0 3d8#0000000080c53030
(1484874087.46) can0 3d8#0000000080c53030
(1484874088.46) can0 3d8#0000000080c53030
(1484874089.46) can0 3d8#0000000080c53030
(1484874090.46) can0 3d8#0000000080c53030
(1484874091.46) can0 3d8#0000000080c53030
(1484874092.46) can0 3d8#0000000080c53030
(1484874093.46) can0 3d8#0000000080c53030
(1484874094.46) can0 3d8#0000000080c53030
(1484874095.09) can0 3d8#0000000080c53830
(1484874095.33) can0 3d8#0000000080c56030
(1484874095.46) can0 3d8#0000000080c56030
(1484874096.47) can0 3d8#0000000080c56030
(1484874097.46) can0 3d8#0000000080c56030
(1484874098.46) can0 3d8#0000000080c56030
(1484874099.46) can0 3d8#0000000080c56030
(1484874100.46) can0 3d8#0000000080c56030
(1484874100.85) can0 3d8#0000000080c53830
(1484874101.46) can0 3d8#0000000080c53830
(1484874102.46) can0 3d8#0000000080c53830
(1484874103.46) can0 3d8#0000000080c53830
(1484874104.46) can0 3d8#0000000080c53830
(1484874105.46) can0 3d8#0000000080c53830
(1484874106.46) can0 3d8#0000000080c53830
[... 129 lines cut ...]
(1484874206.47) can0 3d8#0000000080446030
(1484874207.47) can0 3d8#0000000080446030
(1484874208.47) can0 3d8#0000000080446030
(1484874209.47) can0 3d8#0000000080446030
(1484874210.47) can0 3d8#0000000080446030
(1484874211.47) can0 3d8#0000000080446030
(1484874212.47) can0 3d8#0000000080446030
(1484874213.47) can0 3d8#0000000080446030
(1484874214.47) can0 3d8#0000000080446030
(1484874215.47) can0 3d8#0000000080446030
(1484874216.47) can0 3d8#0000000080446030
(1484874217.47) can0 3d8#0000000080446030
(1484874218.47) can0 3d8#0000000080446030
(1484874219.47) can0 3d8#0000000080446030
(1484874220.47) can0 3d8#0000000080446030
(1484874221.47) can0 3d8#0000000080446030
(1484874222.47) can0 3d8#0000000080446030
(1484874223.47) can0 3d8#0000000080446030
(1484874224.47) can0 3d8#0000000080446030
(1484874225.47) can0 3d8#0000000080446030
(1484874226.47) can0 3d8#0000000080446030
(1484874227.47) can0 3d8#0000000080446030
(1484874228.47) can0 3d8#0000000080446030
(1484874229.47) can0 3d8#0000000080446030
```

adam-nelson, 2017-01-22:
What the latest update?

ingenieroariel, 2017-01-22:
I figured out for ACC which frames report the steering wheel buttons are pressed (0x91) and the one that reports ACC being enabled, the brake or gas peal being pressed and which speed the ACC is set to (0x165).

For LKAS this one reports if it is on or not (0x3d8) and also seems to report info from other sensors, but I need to wait until I drive with a sidekick to learn more.

The only frame I have been able to spoof is (0x82) to interact with the arrow buttons on the steering wheel. Will keep at it and report back.

I am doing all this with an openxc adapter spitting out raw can encoded as JSON conected to my Macbook 12". There are several ways I can use to get data faster:

0. Continue compiling the openxc adapter to ignore a lot of frames and only let few of them pass through. This has been what helped me make progress and the openxc-dashboard and openxc-dump loose a lot of packets when too much info is sent.
1. Switch my openxc adapter to protobuf or messagepack.
2. Wait for the mini cantact I ordered, that one will connect directly to the linux kernel as a SocketCan adapter.
3. Wait for the NEO that's coming from San Diego
4. Build a PCB locally and put the NEO parts I already have.
5. Make the Kvaser Leaf Light v2 I have work with SocketCan.
6. Fly an experienced car hacker, his laptop and his can adapter to Colombia for a couple of days (I've actually offered this to a couple of people but no luck yet :) )

TheMutley, 2017-01-22:
if you have a kvaser interface why are you still using the open-xc interface ? Option number 5 is probably your best bet for reverse engineering you can't afford more expensive tools.

ingenieroariel, 2017-01-22:
It has been mainly logistics. Socketcan does not work on OSX and that's the only laptop I own.

I'll install linux on my laptop and try to get the kvaser working this week, thanks for the hint. When I first tried it I could not get `candump` to work with it (kvaser driver vs linux built in driver, no termination resistor on my cable, etc)

adam-nelson, 2017-01-23:
From what I've seen ford supports the OpenXC adaptor

adam-nelson, 2017-01-23:
@geohot Have you thought about hiring yourself/team out for a fee?

geohot, 2017-01-23:
We are open to people funding ports

ToddGore, 2017-01-25:
I've been a little hesitant to post about can bus codes, but here goes. I also have a Ford Fusion. It's a 2013. It has the parking assist and lane keep assist. 
[image]

[image]

[image]

ToddGore, 2017-01-25:
That connector is for the lane keep assist camera.

ToddGore, 2017-01-25:
I also able to find a program that converts .dlc files to Excel. I converted the ADAS_Kit_20161206 DataspeedByWire.dbc to this. 
[DataspeedByWire.xlsx](https://github.com/commaai/openpilot/files/731212/DataspeedByWire.xlsx)

ToddGore, 2017-01-25:
[image]
[image]

ToddGore, 2017-01-25:
[image]
Still working on this.

ingenieroariel, 2017-01-26:
That excel file is very good to have a layman understanding of each of the dbc rows.

Your connector does not seem to have the 1.7v and 3.3v lines I found on my car but the others are in the same place.

The 0x3d8 command does the same thing mine does. Thanks for sharing Todd!

-a

ToddGore, 2017-01-26:
The .dbc tool that I used is called CANTool. http://www.ingenieurbuerobecker.de/products.html
It's works with excel. Be sure to use the BETA version 4. Also the website is in German! Google translate to the rescue.

ingenieroariel, 2017-02-02:
Another update,

We finally got the Kvaser Leaf Light v2 to work (after adding a 120 ohm resistor and getting rid of the proprietary drivers in exchange for the ones built in latest linux kernel). Thanks @TheMutley 

Instead of going straight for the DBC, we are using a python dict as defined by Canard/Pyvit and will convert to DBC later if needed.

For now the goal is to create a small python project that speaks socket can on one end and talks capnp on the other end as a 'remote' `boardd` running on a cheap dell laptop + the kvaser and ubuntu 16.04.

https://github.com/autti/veloz/blob/master/filterlog.py
```
DECODER = {
     "messages": [
         {
         "name": "Steering Report",
         "id": "0x3A8",
         "signals": {
                     "16": {"name": "Steering Angle", "bit_length": 16},
                    }
         }
     ]
 }
```

Unfortunately pyvit is GPLv3  licensed but we asked nicely if some parts of it could be relicensed, in any case that is not relevant to openpilot as our patch when we finish will be a clean DBC and the fusion implementation.

If anyone wants to collaborate just join Udacity's ND013 slack and go to the #abraham channel.

ingenieroariel, 2017-02-02:
This is the same thing using capnp with the `car.capnp` found on OpenPilot's repo.

Focusing on `CarState` first, then we'll work on the `Sensor` one and finally on the `CarControl`

```( vEgo = 0,
  gas = 0,
  gasPressed = false,
  brake = 0,
  brakePressed = false,
  steeringAngle = 32020,
  steeringTorque = 0,
  steeringPressed = false )
```

The angle is still on the units they provide instead of deg.

eldarial, 2017-02-02:
Ariel, Todd, 

this moves the steering wheel

```sh
cansend can0 415#1A0ED8CE07FB07FB
```

ingenieroariel, 2017-02-12:
Here is a video: https://www.youtube.com/watch?v=QgmAEeaN_b0

And this is the code we used:
https://github.com/autti/veloz/blob/master/send_command.py

ingenieroariel, 2017-02-12:
And here are all the messages we have been able to decode so far: https://docs.google.com/spreadsheets/d/18nHRuRQi4wzUq77X90SLxduJEEQDZcy8s8bXalCvLtw/edit#gid=0

ingenieroariel, 2017-02-15:
After trying different frequencies when sending those messages we discovered a few things:

1. When we replay the messages at a slow speed: the speed meter moves a bit and then steering wheel moves a bit.
2. When we replay the messages at high speed: the speed meter stays at 70km/h and the wheel does not turn, but it is stiff.

This means that the wheel moving with that message was only a happy accident, a side effect of making the car think it was going at high speed was sending a small torque to the steering wheel that caused it to rock a bit. Thanks to @TheMutley for his help asking the right questions.

Good news is that at least we know we can read and write messages fast enough to the car's bus with the hardware we have (cheap dell laptop and the kvaser leaf light v2 with a home made 120 ohm terminator between canh and canl and a DB9 connector connected to the front camera of the car).

Bad news is still we have not decoded the 3 main things we need: gas, brake, steering.

ingenieroariel, 2017-03-12:
Some news, we have been able to make the car accelerate and brake on demand by playing with the `#415 (Speed report)`, if we make the car believe it is going faster, it brakes, if we make the car believe it is going slower, it accelerates.

With this we have been able to isolate a signal that brakes the car but the experiments cause the car to label the ACC (and 7 other systems, hill assist, collision prevent, assisted parking) as not working and I need to reset them via OBD. The painful way has been to actually go to the dealer.

The big change is doing the experiments when driving the car over 30km/h. Reading now about OBD to be able to reset the systems after braking them. If your car complains and disable an specific system because of calls you have been sending, that's a good sign you are onto a juicy command. In our case it's their the front sensors or the actual brake command. Here they are:

0x85
0x186

We have tested on the field that the Lincoln MKZ and the Ford Fusion have the exact same codes. This one moves the steering wheel when in park assist mode:

076

davedave1234, 2017-04-30:
This is great progress, I am 100% new here and am considering a Fusion and if this works out I will definitely buy one. How is it going, anything new to report? 

Thank you for your efforts on this.

alan707, 2017-05-02:
I built a 2016 Ford Fusion instrument cluster (dashboard), body control module, sync module and radio.  All 4 CAN buses are exposed (HS1, HS2, HS3, and MS) so if anyone needs to run a test for a particular CAN message I can reproduce/verify the ID and data on this test bench and report back.

adam-nelson, 2017-05-02:
How did you find all the parts for that?

alan707, 2017-05-02:
Junkyards.  This is the 4th custom test bench I've built so far.  Today I was able to get all CAN buses connected and working for the Fusion so I thought people might be interested.

adam-nelson, 2017-05-02:
how much did it all cost.

I've seen these offical setups at defcon

davedave1234, 2017-05-02:
Awesome, hopefully this will help ingenieroariel find and research what he needs to get this functional.

ingenieroariel, 2017-05-02:
Hey @alan707, thanks for the offer.

0x3D8 is the command that comes from the car's front camera and causes the dashboard to show green, yellow or red.  I'd love to know how the rest of the car reacts to it, in particular if it is directly responsible for waking up another code that applies torque to the steering wheel.

https://github.com/autti/abraham/issues/45
When I start spoofing those my car locks up and declares the module as not working, so all my focus so far has been on creating 'gateway' or 'firewall' modules with dual can controllers.

On the following repo there is one github issue for each code I have seen on my bus, along with tags for how often they appear, any info you can add to them would be great:

https://github.com/autti/abraham/issues/

alan707, 2017-05-02:
OK, I'll take a look - what frequency is the label referring to?

ingenieroariel, 2017-05-02:
That frequency label is how often it appears on a dump from the bus connected to the camera module. We created it to be tell apart common signals from rare ones.

alan707, 2017-05-02:
I sent CAN Message ID 0x3D8 every 10 ms and it did exactly what the description on [issue 45](https://github.com/autti/abraham/issues/45) said it was going to do.  I didn't see any other messages pop-up.  However, keep in mind, this is what I currently have connected on CAN High Speed 3 (HS3):

- Sync 3 Module
- Instrument Cluster
- Body Control Module

According to the 2016 Fusion Wiring Diagram, the BCM should actually be connected on the HS1 bus to the Sync Module, but I had to wire it to HS3 just to get it to turn 

Most other components that might react to to a Lane Keep CAN message are either in HS2 or MS (Medium Speed CAN bus).

alan707, 2017-05-02:
Whenever I replay a candump with `canplayer` from can-utils, I get this error:
`sendto: No buffer space available`

Does anyone know how to fix this?

ingenieroariel, 2017-05-02:
Yes, the same happened to me with openxc, cantact and kvaser with the linuxcan drivers. My solution was to switch to kvaser's private driver that blacklists the one Ubuntu configures automatically.

With that change the connection keeps open and I can send whatever I want without the buffer running out of space. The problem is that I have to so gymnastics to pipe things back to cansnifer via a virtual can.

-a

alan707, 2017-05-03:
hopefully they can recommend a solution: https://github.com/linux-can/can-utils/issues/40

alan707, 2017-05-06:
Can someone confirm 0x3A6#000D yields the response on issue [#31](https://github.com/autti/abraham/issues/31).  This is the command I ran:
`cangen can0 -g 10 -I 3A6 -D 000D`

ingenieroariel, 2017-05-09:
I ran the same command on my car (HS2 bus, the one connected to camera) and got nothing, while in parking, reverse and drive. On which bus did you send it?

alan707, 2017-05-09:
When I had all CAN buses open, that was on HS3.   I confirmed that doesn't work through the DLC (OBD-II port).  Now that I have connected all 4 CAN buses to the CAN gateway module, it is acting more like the real system and that message will need to be piped in directly to HS3

alan707, 2017-05-12:
Here's an explanation how you solve SocketCAN's `No buffer space available` error: [https://community.carloop.io/t/socketcan-canplayer-error-sendto-no-buffer-space-available/298](https://community.carloop.io/t/socketcan-canplayer-error-sendto-no-buffer-space-available/298)

ingenieroariel, 2017-06-14:
I was able to capture data from the 3 high speed can buses using 3 x canable.io ($20 usd), flashing them with the candlelight firmware and using a custom gs_usb kernel extension. Attached are 3 million can messages with LKAS and ACC on from the 3 can buses.

I also recorded video with GPS information using a garmin virb, video at 240 fps around the car with 4 gopros and 4K video front facing. Here is an example:

https://www.youtube.com/watch?v=6VeWhmZnWIw

[3can.zip](https://github.com/commaai/openpilot/files/1075779/3can.zip)

adam-nelson, 2017-06-15:
So thats the car driving itself?

ingenieroariel, 2017-06-15:
Parts of the trip yes, but with the built in system that we are trying to replace with openpilot.

-a

adam-nelson, 2017-06-15:
Well this is good news

@geohots should be happy

ingenieroariel, 2017-06-15:
Here is a non-exhaustive list of id's in those 3 can buses, if anyone wants to help out, you can create issues for the id messages that were not in [abraham](https://github.com/autti/abraham/issues) before:

```
041
042
047
048
04A
04B
04C
05A
05C
05D
076
077
078
07D
081
082
083
084
085
091
092
108
109
14C
156
165
166
167
168
169
171
178
179
186
187
188
18A
18B
192
19A
1E6
200
202
204
20A
213
214
215
216
217
220
221
222
223
224
225
227
228
229
22C
230
242
261
262
265
276
277
279
287
290
2A0
2A1
2A4
2AC
2B1
2B4
2C0
2D0
2D1
2D2
2D5
2D6
2D7
2D8
2D9
2DA
2DB
2DC
2DE
2E0
2E3
2E6
2EE
2F1
2F2
2F3
2F4
2F5
2F6
2F7
2F8
2F9
2FD
303
307
30A
30F
313
315
316
317
318
320
321
326
331
332
333
336
33B
346
348
34E
350
353
35B
35E
360
361
362
37B
386
38A
38D
3A6
3A7
3A8
3AA
3AB
3B2
3B3
3B4
3B5
3B6
3B7
3B8
3C3
3C7
3C8
3CA
3CC
3CD
3D0
3D8
3DA
3DE
3E0
3E1
3E2
3E3
3E5
3E6
3E7
3E8
3EA
3EB
3EC
3F0
3F1
3FB
40A
412
415
416
41D
41E
421
422
423
42C
42D
42F
430
431
432
433
434
435
439
43A
43C
43D
43E
447
453
454
455
456
45A
45E
462
463
464
465
466
467
46A
46B
470
4A2
4B0
4D0
51E
581
591
593
595
596
59E
5B2
5B3
5B5
5C6
5E2
```

adam-nelson, 2017-07-11:
Does the fusion have a sync system?

[49 more replies]

closed 2019-06-02 by pd0wm