commaai/openpilot #10: Consider - a new name for this project?

johndpope, 2016-12-01:
Seems like there's a company already doing drone automation software in this space - 
https://twitter.com/OpenPilot

You could curate this - like by voting - perhaps a simple google form? 
Problem with not renaming it is reasonably foreseeable - when people want to extend this software to actually do drone stuff - then the naming will trip people up.

geohot, 2016-12-01:
We considered this, according to their docs the other OpenPIlot was discontinued and renamed to LibrePilot.

http://opwiki.readthedocs.io/en/latest/user_manual/gcs_install_op.html

closed 2016-12-01 by geohot

MAVProxyUser, 2017-02-22:
Oh Come on @geohot  That is a lame excuse... *some of us* are still around and still make use of OpenPIlot hardware (and respect the name)... The project was not "renamed" to LibrePilot, some of the devs forked off and did their own thing. 

What you referenced is NOT the OpenPilot "project docs"... it never was official project documentation. 

I personally still sell OpenPIlot hardware... (including a navigation capable flight / ground vehicle controller)
https://sasquatchlabs.org/product/openpilot-oplink-mini-modem-bundle/

https://sasquatchlabs.org/product/genuine-nano-flight-controller-w-case/