commaai/openpilot #11: Why only 2016 Civic Touring trim?

dustintownsend, 2016-12-01:
I'm curious why the limitation on the 2016 Civic is for the Touring trim? 
Could this work with other trims? Like the EX or EX-T (if equipped with Honda Sensing)?

jjdblast, 2016-12-01:
Also curious about that.

I think it's maybe some difference in Lane Keeping Assist System (LKAS)  between Touring tirm and other trims. Hope someone could explain that or provide more details.

Here's what I find may helpful:
https://answers.yahoo.com/question/index?qid=20100327004453AA7nn8Z
http://www.edmunds.com/honda/civic/2016/st-401588674/features-specs/
http://www.hondamanhattan.com/blog/2016-honda-cr-v-ex-l-vs-touring/

geohot, 2016-12-02:
Should work with none or very minor changes to the fingerprinting for other trims with Honda Sensing.

If you build it for a different civic, happy to help you make it work.

wcwxyz, 2016-12-05:
What about the trims w/o Honda Sensing? Is Honda Sensing a prerequisite here?

Fryelock, 2016-12-06:
Honda Sense adds the Lane Keeping Assist System (LKAS) and Adaptive Cruise Control (ACC) with Low-Speed Follow. I am still learning but am assuming this is what the openpilot software uses so yes Honda Sense seems to be required. 

It looks like Honda is now offering it on any vehicle, for a $1,000 charge. I came across this thread while trying to build the most affordable Civic with Honda Sense and am happy to see geohot's offer of helping to make it work. Thanks!

[Honda-Automobiles-Personalized-Fact-Sheet636166121240912621.pdf](https://github.com/commaai/openpilot/files/634071/Honda-Automobiles-Personalized-Fact-Sheet636166121240912621.pdf)

nesherhh, 2016-12-07:
I have a Honda Civic 2010. It doesn't have LKAS and ACC. So it is not possible to make auto pilot for it. But I was wandering if it is possible to read more data than it is possible with diagnostic interface (OBD2). Like is light is on, is blinker left or right is on. angle of the steering wheel.... Is this information also available on OBD2? Or I have to look for CAN BUS wires to connect directly to bus?

geohot, 2016-12-09:
ODB2 is the CAN bus on many cars. You should be able to read a lot of information off the ODB port in your Civic.

Krishtof-Korda, 2017-02-04:
I just bought the 2016 Civic EX-L, **without** Honda Sensing package. I think it would be really cool to add ACC to this car for my wife. Are there enough messages on the CAN bus to make this happen or have they been disabled on non-sensing vehicles. I really only need an 'apply brake = 0.xx' and 'apply accelerator = 0.xx' message to accomplish what I am trying to do. I am not familiar yet with the CAN bus. What would I need to start sniffing the CAN and see if the appropriate messages are available?

If possible, I plan to build the kit and do some testing. I just wanted to ask those who have experience with this before dropping the coin to build the Comma One. 

Thanks for any help anyone can provide. I really think this would be a fun project and a way I could contribute to expanding capabilities for openpilot on non-sensing cars.

geohot, 2017-02-11:
Readme updated to reflect that all with sensing work.

closed 2017-02-11 by geohot