commaai/openpilot #5: Test suite - Unit tests and/or E2E tests.

bneiluj, 2016-12-01:
There is no test suite. Unit tests or E2E tests. 

A collection of test cases should be implemented in order to show some specified set of behaviours.

geohot, 2016-12-02:
Pull request please!

mgraczyk, 2016-12-03:
We have unit tests which have not yet been open sourced, because we haven't yet set up automatic testing  on Github. We also have an internal debug/validation testbench/environment that we plan to add to openpilot soon.

Thanks for your input.

closed 2016-12-03 by mgraczyk

bneiluj, 2016-12-05:
Ok - fantastic!

-- 
- - - - - - - - - - - - - - - - - -
Julien Bouteloup
Full-Stack developer.
Start-up Entrepreneur
Software and Hardware Engineer
Eth: 0xfF252f0D10AD3396f98F8652716855CB6b80da32
Bitcoin: 12oySBFevMKQXqrGuiNByhAQoYTLxFUgQN
Cell: +44 7851456368 (UK)
linkedin.com/in/jbouteloup/