commaai/openpilot #7: Can this be used with any car?

Osmosis311, 2016-12-01:
Are all the PID's the same across all vehicles?  Can I use this on a Chrysler or a Ford and determine if the seatbelt lamp is on?

VirenMohindra, 2016-12-01:
No, this will only work with the given models.

DannyBoyBroadSword, 2016-12-01:
Most domestics have unencrypted and non-gateway protected CAN-buses. Oddly I can't believe they got the Honda(Acura) to work given that the CAN packets are both encrypted and protected by a gateway(Which works by asking questions--not sniffing data). Porsches and other Germans which also includes Rolls Royce are now using a system called flex-ray that uses fiber but can only be accessed through a dealer port and most likely only shows diagnostic information for techs. I am fascinated by utilizing existing hardware and using it to drive autonomously.

A quick aside about Tesla and the SLOW automotive industry: 

All modern cars today have the capacity to be autonomous, but Tesla was the first to market the feature outwardly most likely because of their sweetheart deal with regulators and CAFE standard officials. Other companies Like Toyota were singled out by NHTSA after they let go of a plant in Freemont California angering the UAW. Oddly enough Tesla grabbed this factory while Toyota was used as a scapegoat against failing GM and their actual issues with the Chevy Impale and Failing Ignition systems. The Toyota scandal was a sham and only happened with 30 floor mats in the car. Toyota when Nolo-contendere and despite their setbacks have come back strong. They want to see people fail before they come in.

annerajb, 2016-12-01:
What hardware is required on the car apart from a unencrypted or non gateway protected can bus.

I assume the steering should have some sort of servo to allow it to move?
Brake pedal sending a can packet for breaking, should be on almost all modern cars correct?

Is there a way to know offhand it the car have the hardware (servo on steering and brake with can packets) without going and sniffing the packets as the readme mentions?

DannyBoyBroadSword, 2016-12-01:
When you say servo it is implied that their is an active electric power steering component with industrial encoders and enough power to completely control the wheel and not just assist a driver. Most cars with advanced safety packages and electric power steering should have this system. I think we should look at toyota/lexus next because they have really good radar/camera sensor packages. @geohot when to we get into DMCA violations with reverse engineering?

Osmosis311, 2016-12-01:
I'm just looking to get relatively basic info like seat belt light on and
doors open for ford E350 ambulance. Would that be doable?

DannyBoyBroadSword, 2016-12-01:
If it is a ford get a can bus analyzer and start sniffing. Basically reverse engineering CAN is you start messing with stuff and you find the address or packet id of the changing event. Takes time and will become difficult when you get into more recent cars. Start with old cars and find the ones that have minimum sensor packages.

matthewleehess, 2016-12-02:
Commenting because I used to run a custom car shop, and I've done work upfitting electronics on ambulances.  

If you're just looking to grab seatbelt status + door status, you're better off just tapping directly into the sensors themselves.  

The doors will have wires in the kick panels that will have continuity with ground while the door is closed, and (+)12V when open.  Same can be said for the wiring harnesses under the seats.

Not true for all vehicles, other makes/models may use CANBUS end-to-end.  But the Ford E-series should still be using the "older" style 12V system.

Note:  When screwing around with the seat sensors and systems, take great care at every step.  That is typically directly integrated with the airbag system, and can cause catastrophic results.

If you still wanna go through CANBUS, go for it.  Just throwing in my $0.02 about the tried-and-true method.

easyjohn, 2016-12-09:
Hi, I just try to understand which hardware I should add to my 1988 car to be able to use this project.
I can buy separately electric power steering components from listed cars and mount them.
I need some sensors and motors under brake and accelerator pedals. But how I need connect them to comma.ai, since I don't have can bus in car? (i.e. - adapt project for any car)

AadiPapp, 2016-12-09:
Hi 
Can I use this to add ADAS abilities to my Mazda6 2009,
Assuming it is missing some of the software and hardware can I supplement them with NEO board and custom APIs with the sensors or add new sensors with new APIs

TK300, 2017-01-04:
@easyjohn I'm trying to figure out the same thing. Got a car from 1986, maybe we are talking about the same model even ;) As we have to put all the necessary component in on our own i wonder if it would be easiert to transfer such parts from existing cars into ours and sniff the input controlls or make it from scratch. I have no idea about the sensors/radar though...

easyjohn, 2017-01-05:
@TK300 I'm digging a little, so for an old cars we got a two way to implement a Can-bus:
- Take a body computer, a bunch of sensors and servos from Honda Civic 2016 and install them in to old car. Connect this body to exiting sensors (while for adequate result some of old sensors should be running thru some converters or even some kind of fake-modules). This is quite "fast" solution and very possible, that this should in generally work "out of the box".
- Second way is a more complicated, and required really lot of skills - develop your own, open-source implementation of can-bus, based of knowledge about some already grabbed codes (from honda, for example), that not required all sensors and connections to be exists, only that codes, that required for openpilot to work. You still need a power steering and brake/thrust engines/sensors from already existing car model.
A lot of projects was started over the globe to implement open-source can-bus, but as I find, maximum available implementation for now - only can track or generate signals for light/sidemarks/horn and so on...

marcregan, 2017-01-10:
I'm wondering if my 2016 Volkswagen GTI with the is a viable candidate... It has ACC & LKAS. I've found very little online about reverse engineering the CAN-buses on this car. Does anyone know if Audi / VW encrypts their buses?

geohot, 2017-01-10:
No CAN busses are encrypted afaik. Build a NEO and connect it!

Osmosis311, 2017-02-07:
Is there anyone I can hire to help me with this project?  Need seatbelt lamp status for Ford E series.

chintan09, 2017-02-14:
@Osmosis311 I could help you with reverse engineering Ford E.

Osmosis311, 2017-02-15:
@chintan09 that would be great!  can you email me?  danny.ackerman@gmail.com

thanks!

unverciftci, 2017-02-22:
What about Toyota Yaris Hybrid 2013?

ingenieroariel, 2017-03-12:
The answer to the original question is Yes to any car that comes with LKAS and ACC.

Since the conversation later went off-topic, can it be closed?

AadiPapp, 2017-03-12:
Hi Ariel,

How about cars with no LKAS and ACC.
I think the topic is being forced on this conversation.I see no harm in
keeping this issue open.
The open ended topic hope helps to come up with generic solution for all
cars.

Thank you
Adi

ingenieroariel, 2017-03-12:
(disclaimer I don't have any affiliation with Comma)

For those cars with no LKAS or ACC I believe the recently released Panda board will be a start. And then solutions like OSCC onceit is more mature and addresses the safety concerns with things like steering torque.

About keeping it open or closed, that's up to the maintainers, I agree the conversarion about generic solutions needs to continue, here or elsewhere.

AadiPapp, 2017-03-13:
@Ariel Thank you and totally agree with you.Looking fwd to George's talk on
Panda board at SXSW :)

ajarrard, 2017-05-03:
@easyjohn @TK300 Have you all found out anymore on this issue? I too would like to bypass the car's system and adapt this for use on an older car by adding my own motors and sensors.

robertcw-3rd, 2017-10-17:
So I’ve got a 96 Chevy Camaro z28 that I’d love to get auto cruse control / whatever it’s called to have the car auto follow the traffic in front of me.  This would be outstanding in Orange County and LA traffic, especially on 405.  

The car did come with cruise control from the factory, but it’s a 90’s Chevy, so I doubt this will be possible.

QTRBENZStageIIautDvO, 2018-03-20:
Hello everyone. I have a lot of questions. But first off, Im new to the idea that has been eating at me for a very long time. I feel like everything I've seen, read and believe to understand- is I can make my commuter a possible level 2 driving car with this program. If anyone is still here. ( been 4 months since a comment so I feel this is still fresh. ) could and would anyone help me with some serious questions concerning my vehicle. Thank you very much in advanced!

bugsy924, 2018-03-20:
https://github.com/commaai/openpilot#supported-cars

closed 2018-12-18 by vanillagorillaa