commaai/openpilot #12: False (?) statement about Open (!) Source?

msva, 2016-12-02:
Hi!

You state that
> openpilot is an open source driving agent
in project's README file.

That's good and very, very admirable.

But...
let's take a look on https://github.com/commaai/openpilot/blob/master/selfdrive/visiond/visiond
What we will see? A blob. 14Mb (!!!) blob!
Also, let's take a look on https://github.com/commaai/openpilot/blob/master/phonelibs/hierarchy/lib/_hierarchy.so
Also 0.75M blob (at least, not 15Mb).

And also `board/tools/dfu-util*`:
1) blobs too (and confidence is doubtful at this point, after two hidden blobs)
2) almost ***every*** distro has `dfu-utils` available in package managers.
3) there is much more CPU architectures than x86_64 and aarch64 (and even talking about arm-related arches: I can't count aarch64 boards with fingers of only single hand. Most ARM hardware (even that one that cosplays desktops) is still 32bit.

geohot, 2016-12-02:
_hierarchy is just scipy's hierarchy, already open source, we can provide the version etc...if you want to rebuild.

dfu-util is open source elsewhere, it's the same tool. those binaries are just for convenience.

visiond is closed source for now, this has been addressed in a previous issue. Just like how certain parts of Android are closed source.

Don't hate. Too many exclamation points for a dispassionate inquiry.

geohot, 2016-12-02:
hierarchy: https://github.com/scipy/scipy
dfu-util: https://github.com/dsigma/dfu-util

closed 2016-12-02 by geohot

msva, 2016-12-02:
1) I didn't state that either of dfu-util or scipy is non-opensource. Just you distribute self-compiled blobs (that can or can not contain extra code, that is executed under special circumstances, and it is pretty hard to prove either it's existance or non-existance).
That's why git has submodules and make (and other buildsystems) has support for recursive builds and dependencies.

So, ideally, there should either be sources (which will be build altogether with the whole project, but ***not*** prebuilt), or they should be stated as build (and/or runtime) dependencies.

Shipping precompiled binaries is ***not*** an OpenSource way. As I already said, it is pretty hard to prove that they're clean from "extra" functions, that is not in upstream code.

2) exclamation marks doesn't mark hate. They mark places which requires extra attention.

3) 

a)
> visiond is closed source for now
I hope, that it means "it will be opened too in near future". Otherwise, as I stated, shipping blobs in project claimed itself as "opensource" is, *technically*, a kind of lie (simple logic: open source is when source is open. If it is closed, even partially, it is not open).

b)
> this has been addressed in a previous issue
I'd say "ok", but, technically it is also not fully true.
First of all, all the answers about that was, let's be honest (yeah, I'm a CEO in IT company and I know the real meaning of such words about a reason of keeping `visiond` closed in that issue), is... uhm... let's say, not as true, as pretended to be (if talking without rude words).

c)
> Just like how certain parts of Android are closed source.
Nope. AOSP (the thing you're referenced as android) is fully useable without a single closed-source blob, if you'll run it on hardware which does not require proprietary drivers (or, say, emulator).
Only closed-source things is 3-party drivers (for 3-party hardware, that Google doesn't produce by themselves) and applications (which is totally not required for running AOSP (android) as OS).

And, anyway, main issue is in that fact, that until you ship closed-source blobs as part of the project, it is (attention, please) ***NOT*** an OpenSource, but just false advertising (sic!).

geohot, 2016-12-02:
Feel free to submit a pull request to build dfu-util and hierarchy from source.