commaai/openpilot #18: Open ?

gameinskysky, 2016-12-05:
We can not modify anything, not really open.

mgraczyk, 2016-12-05:
What would you like to modify? You are welcome to submit a pull request.

penne12, 2016-12-05:
@gameinskysky That's not how open source works.... open source means that the source is available under a license that allows you to modify anything and publish your modifications, not that you can modify anything in the main version of the code. 

Here's a link to the license. It's pretty much a standard MIT license.
https://github.com/commaai/openpilot/blob/master/LICENSE.openpilot

Or, for a summary, may not be completely accurate:
https://tldrlegal.com/license/mit-license

If you click ![Fork button][http://bsite.cc/iLDC/Screen%20Shot%202016-12-05%20at%209.38.16%20AM.png] (next the top right corner of the site), you're able to create a copy of this project that you can modify. Then, you can submit a pull request requesting that @mgraczyk (or another collaborator) accepts your changes.

closed 2016-12-06 by mgraczyk