commaai/openpilot #20: Add linuxcan to NEOS kernel

ingenieroariel, 2016-12-06:
Being able to access SocketCAN from the phone would make it possible to connect a KVASER adapter or any other CAN adapter with linux support on the linux kernel over USB.

 - [ ] can
 - [ ] can_dev
 - [ ] can_raw
 - [ ] kvsaser_usb

I would also help during reverse engineering as the candump output can be relied back to the computer during debugging.

paramaggarwal, 2016-12-12:
Very novice comment here - but I think the way it goes via an intermediary right now isolates it from software failures on the phone as well as places reasonable limits on what the phone asks the CAN bus to do. Hence connecting directly to CAN over USB via a transceiver would basically hand over the controls over the car to the phone - exposing it to typical Android vulnerabilities.

geohot, 2016-12-12:
Yea, the board adds one layer of safety redundancy, and that's one of the reasons we built our own, although you can achieve very similar things in software. The comma zero used off the shelf CAN transceivers safely by adding the safety checks to the drivers.

ingenieroariel, 2017-03-12:
Even if the support was added, we would need Python 3 on the OP3 to be able to connect to the socket (socket can did not work for me on 2.7).

Also, since the NEOS kernel is not on this repo but a new github repo I'll close this.

closed 2017-03-12 by ingenieroariel