commaai/openpilot #21: Publish openpilot on PyPi

ingenieroariel, 2016-12-06:
So end users can just do:

```
pip install openpilot
python -m selfdrive.manager
```

geohot, 2016-12-06:
Looking in to it.

ingenieroariel, 2016-12-06:
```
import os
from distutils.core import setup
from setuptools import find_packages

def read(*rnames):
    with open(os.path.join(os.path.dirname(__file__), *rnames)) as ff:
        return ff.read()

setup(
    name='openpilot',
    version='0.1',
    author='',
    author_email='',
    url='https://github.com/commaai/openpilot',
    description='openpilot by commaai',
    long_description=(read('README.md')),
    classifiers=[
        'Development Status :: 1 - Planning',
    ],
    license="MIT",
    keywords="commaai openpilot",
    packages=['cereal',
              'openpilot',
              'selfdrive'],
    include_package_data=True,
    install_requires=[
        'pyzmq',
        'pycapnp',
        'raven',
        'setproctitle',
        'libusb1',
    ]
)```
I have a branch where I did the legwork to make it happen:

https://github.com/autti/openpilot/tree/flying-high

If the approach is sound I can submit a PR.

Note: I also took the liberty of reserving the name on pypi before somebody beats us to it. I can transfer ownership to @espes or anyone you indicate.

NeoFromMatrix, 2017-05-06:
Seems that an initial version might be pushed, is there anything further planned?

closed 2019-06-02 by pd0wm