commaai/openpilot #22: Allow setting daemons via environment variables

ingenieroariel, 2016-12-06:
Right now we have to comment code in manager.py in order to disable a service, except for logging for no board.

If we move all daemons to environment variables we should be able to disable them with envvars and to point to third party implementations of each daemon:

Here is an example:
```# Daemons usually start as python modules and  graduate
# to become c++ services, like what happened with boardd.
# In order to support a seamless transition and let end users
# override the services now all deamons read from an environment variable.
# The possible values are:
# 1. A fully qualified python module, like 'selfdrive.controls.radard'.
# 2. A path to an executable, like '/data/openpilot/visiond'.
#       * If the module is not build, this application will try to build it
#         from a Makefile in the parent folder.
# 3. An empty string, which disables the module.
#
# Any value that does not satisfy any of the conditions above will trigger an Exception
#   and abort execution.

managed_processes = {
  "uploader": os.getenv("OPENPILOT_UPLOADERD", "selfdrive.loggerd.uploader"),
  "controlsd": os.getenv("OPENPILOT_CONTROLSD", "selfdrive.controls.controlsd"),
  "radard": os.getenv("OPENPILOT_RADARD", "selfdrive.controls.radard"),
  "calibrationd": os.getenv("OPENPILOT_CALIBRATIOND", "selfdrive.calibrationd.calibrationd"),
  "loggerd": os.getenv("OPENPILOT_LOGGERD", "selfdrive.loggerd.loggerd"),
  "logmessaged": os.getenv("OPENPILOT_LOGMESSAGED", "selfdrive.logmessaged"),
  "logcatd": os.getenv("OPENPILOT_LOGCATD", "./logcatd"),
  "boardd": os.getenv("OPENPILOT_BOARDD", "./boardd"),
  "ui": os.getenv("OPENPILOT_UI", "./ui"),
  "visiond": os.getenv("OPENPILOT_VISIOND", "./visiond"),
  "sensord": os.getenv("OPENPILOT_SENSORD", "./sensord"), }
```

mgraczyk, 2016-12-16:
Instead of this many environment variables, it would probably be more appropriate to use a config file. We pass around a globally shared immutable context called `gctx` throughout the code. We could include these values and load `gctx` from a file in manager_init. @geohot what do you think?

ingenieroariel, 2017-03-12:
Has this been addressed in later internal releases?

I have my own `boardd` and would love to just pass the right configuration to swap it.

geohot, 2017-03-12:
Internally we just edit manager. We have a few internal services now too, maybe we'll have an external file with the service list in 0.3

espes, 2017-06-15:
This can now be done with manager_extensions.py

closed 2017-06-15 by espes