commaai/openpilot #23: How do I make this work for Toyota Rav4 2016

nojvek, 2016-12-07:
It has adaptive cruise, 6 radars, 6 cameras and lane detection already.  I would love to make it work with open pilot.

geohot, 2016-12-07:
Would love to support this car, and I think it's a very reasonable next target.

You need to reverse engineer how the LKAS and ACC are being commanded. I would get access to TIS, the Toyota Information System, that has schematics for that car. Then tap the CAN bus that the forward camera sits on and sniff it.

nojvek, 2016-12-07:
Thanks for the reply. I don't consider myself a CAN bus hacker but its always something I really wanted to learn.

https://techinfo.toyota.com/techInfoPortal/appmanager/t3/ti?_pageLabel=ti_whats_tis&_nfpb=true

I'll try to get a 2 day subscription and dabble my head into it.

I have made it a goal of 2017 to make my car follow lanes automatically.  

I just hope this doesn't violate the lease.

spant, 2016-12-07:
Could you just check which cars available in India could be a target for this?

geohot, 2016-12-07:
We will be releasing an updated version of openpilot soon with a car abstraction layer, showing the exact list of things you need to find.

But assume you will definitely need the gas, brake, and steering control packets. Best way, tap forward camera CAN, engage ACC and LKAS, and figure out which packet did it.

nojvek, 2016-12-07:
Sounds great. I'll keep an eye out and purchase the CAN related equipment.

I was always under assumption that the bus would be heavily encrypted so no
one could tamper with it.

What you're saying is, it's basically frames as C structs that need to be
decoded right?

geohot, 2016-12-07:
Oh no, not encrypted at all. You can also use a comma neo board to interface to CAN, then boardd from openpilot to receive CAN frames. The encoding for the Honda is in the DBC files.

nojvek, 2016-12-07:
Took a look at the dbc files.
https://github.com/commaai/openpilot/blob/master/dbcs/acura_ilx_2016_can.dbc

Is the format documented anywhere? Trying to make sense of the different
fields.

The comma team painstakingly reverse engineered and built this files right?

Once I get the neo equipment, I'll see how far I go and get back to you.

jon-mullen, 2016-12-07:
Looks like it didn't come from a manufacturer, they usually have a list of modules and some versioning stuff in there.

Might want to check out https://github.com/ebroecker/canmatrix.

Hopefully a couple examples help:

### Messages:
#### BO_ 0x0E4 STEERING_CONTROL: 5 ADAS
(BO_ = following is a message)
(0x0E4 = message number in hex)
(STEERING_CONTROL = message name)
(5 = message length in bytes)
(ADAS = message sender module name)

### Each signal:
#### SG_ STEER_TORQUE : 0|16@1- (1,0) [-3840|3840] "" Vector__XXX
(SG_ = following is signal)
(STEER_TORQUE = signal name)
(0 = signal starting bit)
(16 = signal length in bits)
(1- = treat those 16 bits as a twos-complement signed integer)
(1 = multiply signed integer by 1)
(0 = then add zero)
(-3840|3840 = resulting range)
("" = no unit)
(Vector_XXX = receiving module for this signal)

### One more signal from a different msg, this one's unsigned with a nontrivial factor:
#### SG_ XMISSION_SPEED : 0|16@1+ (0.002759506,0) [0|70] "m/s" Vector__XXX
(SG_ = following is signal)
(XMISSION_SPEED = signal name)
(0 = signal starting bit)
(16 = signal length in bits)
(1+ = treat those 16 bits as unsigned integer)
(0.002759506 = multiply unsigned integer by 1)
(0 = then add zero)
(0|70 = resulting range)
("m/s" = unit)
(Vector_XXX = receiving module for this signal)

The value tables (VAL_) at the bottom spell out what the values mean for the state-encoded signals.

nojvek, 2016-12-07:
Great. This is useful.
https://github.com/ebroecker/canmatrix/blob/master/canmatrix/dbc.py

> Looks like it didn't come from a manufacturer

Is this something that manufacturers usually publish in their information
system that could be accessed by a subscription?

Regards.

geohot, 2016-12-07:
No, we had to reverse engineer it. Good luck finding them

Mutley2017, 2016-12-07:
The DBC files are Vector's proprietary format for CAN database and that is usally what everybody who is serious about CAN is using (OEM, Tier1 etc..). These files can usually be opened with CANdb++ which is available for free when you download the CANalyzer demo. Now, I found out that the files DBC files generated by CANMatrix when switching from another database format to DBC are not directly compatible with CANdb++. I had to do some modifications to them in order to be able to open them. It's worth it since this software is very useful when constructing your very own DBC.

hableel, 2016-12-12:
I have 2013 Honda accord without LKAS, ACC.

So in order to control steering the car must have LKAS ?

What about brakes. Standerd cruise control disengage when braking.

Is the functionality of controlling steering and brakes limited by turning ACC,lkas on or that just the way neo designed?

nojvek, 2016-12-13:
@hableel, NEO is basically going to send steering angle, acceleration and
brake commands to the car's CAN bus. The car needs to be able to understand
those commands.

If the car is not supported by openpilot then this means you'll have to
hook into the correct bus and reverse engineer the commands your self.

hableel, 2016-12-13:
@nojvek cars equipped with necessary systems will understand command because its just sniffed from it. but will it accept commands when turning off lkas,,acc ? other cars that don't have this functionality i don't what they can do because they are not designed to change steering angle. 

My car change steering resistance based on speed and brakes. maybe there is a way to control steering angle as well but I'm not sure

adam-nelson, 2016-12-15:
Thanks @geohot for OOS the project. I myself aren't a coder and have an interest in improving self driving or driver assisted technology. Because I am geek and want to understand

How friendly is ford when comes to reverse engineering cars from 2010 onwards?

Whats the simple checklist for car suitability for self driving or assistance and can I create aftermarket parts that add features to the car.

nojvek, 2016-12-15:
I look forward to day when cars expose, camera, radars, throttle and
steering over a common protocol.

My canbus to USB has arrived by mail, I'll get back on how the reverse
engineering goes.

rmueller83, 2016-12-16:
Perhaps this will help you to get started
[http://illmatics.com/car_hacking.pdf](url)
[http://illmatics.com/car_hacking_poories.pdf](url)
[http://illmatics.com/can%20message%20injection.pdf](url)

adam-nelson, 2016-12-16:
I think you also need to packet sniff a car

olmerg, 2017-03-29:
Hi, I am reading this thread and I want to recommend this can analyzer software [Busmaster](https://github.com/rbei-etas/busmaster) it use dbf files and personnally is easier than Vector programs.

closed 2019-03-12 by vanillagorillaa

andrelondono, 2021-12-23:
Idk i want to do it with 14 cameras on a 2015 honda accord noot