commaai/openpilot #24: cellular network is not working after flash install

xinlutang, 2016-12-09:
Get a fresh Oneplus 3 phone, and test the sim card working before I unlocked and flash install to Commaai binary. The installation went well without an issue, but the cellular network is not working.  The continue.sh can not git pull from this repository. After connected to the phone with connect.sh, I can not ping google.com. 

I then took out the SIM card and tested on other regular phone to confirm the problem is not from SIM card itself. I suspect there is something broken the the system image in this release to set up the cellular network.

geohot, 2016-12-09:
What network and country? Can try to test.

xinlutang, 2016-12-10:
US， ATT

ingenieroariel, 2016-12-10:
I tried the same in US on ATT and it worked fine with the same release. I was using sim slot 1.

One thing that could help users in this situation is to not block boot if sim card is not present - and let them choose to connect over wifi from the UI.

ingenieroariel, 2016-12-10:
@xinlutang Which bands does your phone support? One test you can try is to flash with stock firmware again and check if the cellular network is picked up correctly, that'd rule out any compatibility problems with your hardware.

andrewraharjo, 2016-12-19:
@xinlutang for ATT you might want to check https://www.att.com/esupport/article.html#!/wireless/KM1062162

before you start flashing OnePlus3. I figured set up the Oxygen OS with the correct APN and test it out will let you keep going run the continue.sh script. Here's the step how you can debug it;

Oxygen OS
1. Insert SIM Card
2. Setup ATT APN for One Plus 3
3. Get in to the dev mode --> enable OEM Unlock and USB Debug
4.  run ./unlock.sh
5. run ./flash.sh
6. run the installation after NEOS boot (copy the rsa and continue.sh)
7. Run the ssh connect.sh
8. cd .. on the termux and run the continue.sh

Before you run the continue.sh check whether your APN allows you to ping out

Hope it helps

geohot, 2016-12-20:
hmm, so flash will wipe any APN you set, you may have to set the APN from the command line in NEOS

andrewraharjo, 2016-12-20:
I thought the same thing. Tried wpa_cli, can't do scan either add_network. Super frustrated at the beginning. One trick I did, I didn't lock after run ./flash because it says optional. I remember before flashing it I did setup the APN correctly on Oxygen OS.

geohot, 2016-12-29:
I'm not sure if the APN settings menu is compiled in to NEOS. If it is, you should be able to get to it from the command line with "am start"

closed 2019-04-09 by legonigel