commaai/openpilot #26: Safety, Ethics and OpenPilot

5f4uw3Uj9gcHuPP8, 2016-12-09:
This post addresses a project that is under active development. This issue will lead to pull requests which make changes to the code in order to address some of the forthcoming concerns. That being said, the questions I am asking reach much further than changes to the source code. 

I am very happy to see autonomous driving code in the public domain. This technology will radically change our future; the more people who can work with and understand these concepts, the better. Recently @geohot gave a very motivating speech at a Udacity event. However, it was shocking to hear George claim that he had not heard any critical opinions regarding the safety of his plan. "His plan" being that Comma.ai provide the source code of OpenPilot while individuals assemble their own Comma.ai NEOs, modify the source code to hack into their particular cars, and run OpenPilot while driving on public roads. In my opinion this plan is dangerous, and needlessly so. I hope that the community can come to a place where we maintain all the social goods of this open source research opportunity without taking any unnecessary risks. This is especially important because the cost of failure is so high. This is not like hacking your PS4 where the worst outcome is that you might brick your system. We are talking about possible fatalities here; fatalities which are completely avoidable.

Some Safety Risks:
* **Constant collaboration may prevent anyone from fully understanding the system** While the end user has access to the source code it is unreasonable to expect that they will understand the edge cases and nuanced behaviors that come from the interactions of its many parts. This is especially true if OpenPilot is developed by dozens of people simultaneously in a distributed manner. 
* **Failure modes are poorly communicated and understood** Keep in mind that control of the vehicle is achieved by hacking into the vehicle's subsystems such as the Cruise Control or Lane Keep Assist functionality. The documentation for the _nominal_ behavior of these subsystems is not released by manufactures. The behavior of these systems in the presence of a fault or bug in OpenPilot is even less clear.
* **Driver assistance systems introduce risks which are not currently addressed** @geohot 's stated primary goal in releasing this project is to see the community provide implementations for vehicles beyond those hacked into by Comma.ai. The work required to hack into a car can vary greatly. Some systems use proprietary commands, and even those that are standardized may interpret commands differently. Many systems have different hard-coded limitations. For example @geohot has commented that there are different minimum speed requirements to activate Cruise Control among the cars he has personally worked on. It stands to reason that crowdsourcing is one economical way for a small startup to be interoperable with a wide variety of vehicles. The danger is that some unknown aspect of these systems will only be activated in rare circumstances. Because there has been no thorough systems analysis and no collaboration with the OEMs there is no mechanism to anticipate these rare events. Due to the development process that is being proposed for OpenPilot, many of these rare events will happen for the first time on the road, perhaps with deadly consequences.
* **Unknown capabilities of the perception system prevent users from making informed decisions** Openpilot fundamentally relies on the "visiond" system to provide the path that the car tries to follow. "visiond" is a closed source executable which relies on machine learning to function. It is not clear what data was used to train this module. Therefore, the end user can not form a reasonable expectation of what the car should understand! Furthermore, @geohot has openly admitted there are certain road markings that the car can not navigate well (such as merging lanes) but no exhaustive list of such limitations has been communicated, and perhaps not even created by Comma.ai.
* **Huge liability for OpenPilot users** While the strategy of not selling anything allows Comma.ai to avoid regulation it does so at the expense of all the people who might contribute to this project. I am not a laywer, but I think any reasonable person can see how dangerous it is to use the Comma.ai NEO on public roads. In the event of a collision the driver of the vehicle using OpenPilot would almost certainly face legal scrutiny that they could not stand up to. Collisions aside, hacking into your vehicle is very likely to void your warranty. People might be willing to accept these monetary/legal risks, but it should be made extremely explicit to the contributors. 
*  ** Some individuals _will_ cause a hazard with this code** Some users will be experts who make valuable contributions, and they might even be hired by Comma.ai. Most people who download this project will not have the interest or the skill to use it. But somewhere in the distribution of skill levels there are people who are dangerous. They are skilled enough to follow the directions and get everything running, but not skilled enough to really understand what they are doing. Those people are usually kept away from safety critical applications through hiring processes or by being supervised by more senior individuals. The current development strategy is subjecting the public to developers who will take wild risks perhaps intentionally, but certainly unintentionally.

Autonomous vehicles hold great promise to reduce fatalities and enable fundamentally new and better ways to live our lives. There may be some unavoidable risks on the path to that better future, but there are many more avoidable risks. The above points are just a few examples of avoidable risks. To make an exhaustive list is beyond me, but I think it is something we as a community must create. And when we have that list of avoidable risks in hand, it is our duty as ethical developers to attack that list until there is nothing left on it.

geohot, 2016-12-09:
I appreciate you engaging about these topics. Would prefer you did so with a real world identity and without a throwaway account though.

This is not a product for end users. As stated in the README, this is open source software for research purposes only. See https://github.com/CPFL/Autoware and https://github.com/PolySync/OSCC for examples of similar software.

Re: failure modes of closed source cars, this is a concern we have. I would love to see pressure on auto manufacturers to open this information up.

Re: visiond being closed, I address this in the presentation. Consider it a QM, https://en.wikipedia.org/wiki/Automotive_Safety_Integrity_Level#QM part. I believe the Mobileye EyeQ3, the chip Tesla uses, is QM as well. All code actually controlling the car is open source, and that's where the safety checks lie.

Re: the legal status of openpilot, comma.ai does not warranty this software for any purpose. People downloading this code are responsible for using it in a safe and legal way in whatever jurisdiction they may lie. We are not selling or shipping a product, and you are responsible for using this software safely and responsibly.

Re: some individuals will cause a hazard. This can be said about many things from kitchen knives to poorly assembled furniture. As with the two previous examples, which to date have killed way more people than self driving car testing, I believe most people will use this responsibly.

Bottom line, cars are dangerous. Inaction is still an action.

toxicgumbo, 2016-12-10:
I wasn't the original writer, but your response is a good indicator that you're distancing yourself and Comma.ai from responsibility.  I hear and read increasing attempts to disconnect from accountability and insinuate that anyone who gets hurt brings it upon themselves.  If anyone is being vague or unclear, I'll say it plainly: You are naive, reckless, and will kill people through misleading them.  Rather then defend yourself, try actually responding to what people are asking for.  Build a subgroup.  Asking for pull requests is simply a way to delay, deflect focus, and demand that others clean up the Proprietary Pilot mess.  Put up a cash bounty if you want to play that way.

You trumpet this science experiment as if it's superior to everything on the market (except perhaps Tesla).  You use it illegally on public roads.  You avoid government safety legislation.  You keep referring to your device as "open source" when only a subset is (don't fool yourself into thinking QM applies here).  You build up enthusiasm and market yourself freely and widely in the press, badmouthing those who are working aggressively to use and develop safety features, redundancy, and functionality at the risk of visible progress.  That's a cowardly shot for you to take because the bulk of people aren't keeping up with what these companies are doing.  Meanwhile, you're claiming that a crappy cellphone and rehashed CAN code for a narrow range of pre-existing driver-assist hardware replaces all of this.  People are get excited and shouldn't be.

People kill themselves with kitchen knives and poorly assembled furniture.   It's a poor comparison.  Those aren't autonomous or rely on artificial decision making designed by someone else.  

Bottom line, you will kill people with a product the government pleaded you not to distribute and that a growing number of people are seeing for what it is: a cobbled together hack of Instructables.

geohot, 2016-12-10:
Huh? Proprietary Pilot? Cash bounty? Don't fool yourself into thinking QM applies here? Kitchen knives kill people but it's okay because they aren't autonomous? We use it illegally on public roads? (citation needed)  People are get excited and shouldn't be?

The government asked us to delay sales of a product. We did exactly that. As most people seem to understand, this is research code only. Happy to engage constructively, but If this conversation becomes toxic I will close this issue.

toxicgumbo, 2016-12-10:
Citation very publicized.  I don't work for you.  It's out there amongst all the other publicity you've been embracing.  There's plenty of lockerroom talk about how great, perfect, and revolutionary this project is, but little to none about safety and its use only as research.

I read the NHTSA response.  Why apply for advancement of a product based on code considered "research"?  For attention?  For credibility?  To attract developers?

George, if your project means something, stand behind it in all aspects.  Publish how it won't get anyone killed and you're behind that initiative 100%.  Insurers, laws, and automakers identify the autonomous car now as the "driver" and responsibly party.  That means Comma.ai has/will have legal ties to consequences.  You posted two other projects where safety is deliberately written into the design.  It's discussed regularly on the mailing lists and publicly.  Do the same.  Make it (and your users) a priority.

It's good that you're willing to discuss things constructively despite a rephrase of words and threat of closing this issue (fortunately, it's being read and is archived).  I and others have been promoting constructive ideas and I've spoken honestly and openly.  I haven't read anything constructive by you yet, particularly in these areas which seem most concerning to many of us.  Mostly I see an unwillingness to talk genuinely about the issues. 

If closing this issue is an easy way to move forward and live in the fog of fanboys and avoid criticism, no worries.  Others will speak up.

geohot, 2016-12-10:
This is not an autonomous car. This is a repository on GitHub with free software that implements a driver assistance system. We discuss safety @ https://github.com/commaai/openpilot/blob/master/SAFETY.md

I'm closing this issue, if you have specific actionable concerns please open issues for them.

closed 2016-12-10 by geohot

geohot, 2016-12-10:
To the original poster, both the constant updating and car failure modes issues are things we'd like to address. Suggestions? We already had planned to make it more clear when the code updates, and we'll be probably adding a release branch for stability. Re: car failure modes, we've done our best to reverse engineer error codes from the car, but I agree a lot more work needs to be done in this area.

mslavescu, 2016-12-12:
@toxicgumbo unfortunetly this whole project seems to be very limited, with core components closed source, and the founder very unfriendly :-( just look how he closes most of the issues.

If you'll like to contribute to an open initiative not tied to any company profits, that is looking to tackle and solve all problems related with Self Driving Cars, check out this new organization:

http://github.com/ossdc

Join the discussion here:

http://ossdc.org/join

paramaggarwal, 2016-12-12:
I fail to understand the criticism towards this project here. It's free code and you do whatever you like with it. One could either make good things from it or bad. It's for research and exploration and not a product.

By definition of open-source and MIT license, there is no liability of any sort on the creators of this project. Everyone in the industry can learn and take the good parts and ignore the bad parts.

Can we please work towards making cars safer together? And not try to shoot down new projects for lacking complete code release or lack enhanced safety checks. It's not a product, it's not mature.

paramaggarwal, 2016-12-12:
The original poster and follow-up criticism in this thread are not constructive. They cite fear and uncertainty but no concrete suggestions on how to deal with it.

**Safety**:
Human drivers are unsafe and unreliable. Having an additional assist system by definition can only help improve this. Please understand the difference between _safer_ and _safe_. Nothing is perfect - not even your Tesla.

**Open-source**:
Trying to dictate what parts of the stack someone should open is like asking for a free lunch. Yes, some parts of not open. But its the same strategy as an Android phone vs. the Android OSS or the Chrome browser vs. Chromium project.

**Research**:
Personally I have learnt a lot by going through this project - I do not intend to assemble one, but it has been educational to me. 

**Public Roads**:
A discussion about drive-by-wire cars is different from driving these development/research cars on the road. If someone chooses to use someone else's code on the road and put themselves at risk - that is their own fault and very silly of them.

Let's keep the discussion of open-source and research of self driving vehicles on one side and use of said vehicles and corresponding safety on public roads separate. By choosing the merge them, the user is taking the risk in their own hands and should very well be aware of what they are doing. 

This is the same as driving - by choosing to drive the car yourself rather than let someone more experienced do it - you are putting yourself and others at risk.

geohot, 2016-12-12:
Re: constructive criticism. We are always looking for ways to make openpilot safer. We believe that self driving car researchers are safer than other groups of people legally allowed on public roads, namely teenagers, old people, and "buzzed" people.

Superbrainpain, 2016-12-12:
Hi George, I saw your recent presentation at Udacity and thought you would
make a good speaker for Singularity university. It would be a great way to
promote Comma and also I believe they actually pay presenter (nice bonus).

Here is there note back regarding my inquiry...

Exponential Conferences Support
7:01 AM (56 minutes ago)
to me
Hello Robin,

Thank you for your inquiry!

If you would like to submit a speaker request for yourself or another, just
shoot me an email with the following information:

The name of the summit (Exponential Manufacturing, Exponential Finance, SU
Global Summit, Exponential Medicine) that you think the speaker would be a
good fit for.

Please provide a little background on the speaker: personal bio, past
speaking events, company they work for, CV, ect

We will also need name and contact information for the speaker you wish to
nominate. It is also helpful to have a topic and brief description of the
talk.

Once I receive that, I can send it to my Conference Director for review.

Best,
Dave Kutz

SU Conference Support

conferencehelp@singularityu.org

toxicgumbo, 2016-12-12:
The following two comments seem to have gone missing:

26#issuecomment-266348650

> Marius Slavescu <notifications@github.com>
> to Mention, commaai/openpi., me 
> @toxicgumbo unfortunetly this whole project seems to be very limited, with core components closed source, and the founder very unfriendly :-( just look how he closes most of the issues.
> 
> If you'll like to contribute to an open initiative not tied to any company profits, that is looking to tackle and solve all problems related with Self Driving Cars, check out this new organization:
> 
> http://github.com/ossdc
> 
> Join the discussion here:
> 
> http://ossdc.org/join

26#issuecomment-266363034

> Param Aggarwal <notifications@github.com>
> to commaai/openpi., me, Mention 
> I fail to understand the criticism towards this project here. It's free code and you do whatever you like with it. One could either make good things from it or bad. It's for research and exploration and not a product.
> 
> By definition of open-source and MIT license, there is no liability of any sort on the creators of this project. Everyone in the industry can learn and take the good parts and ignore the bad parts.
> 
> Can we please work towards making cars safer together? And not try to shoot down new projects for lacking complete code release or lack enhanced safety checks. It's not a product, it's not mature.

Two important words in this thread: _constructive_ and _research_.

> The original poster and follow-up criticism in this thread are not constructive. They cite fear and uncertainty but no concrete suggestions on how to deal with it.

**Constructive**: Constructive is what one makes of it.  If criticism leads to a dismissal of concerns and closed tickets, then there is nothing constructive going on. If my tone is somehow offensive to others here, my apologies. However, expressing concerns does not require expressing concrete suggestions.  There is genuine fear to be recognized here--people are writing code around closed systems anticipating the movement of 3-ton vehicles in the company of the public--and "how to deal with it" is exactly what should be an ongoing discussion (from day 1).  By putting this code in the public and claiming a reliability level on par or exceeding systems known to be some of the safest around, the public is inherently less safe.  Concrete suggestion: Active and ongoing public discussion of security and safety concerns ranging from foggy weather, to CAN injection, to external electrical interference, to tire blowouts, device hardlocks, optics failures, HMI intrusion (deliberate or otherwise), non-responsive driver, unexpected response from subsystems, etc.  

**Research**: The use of this word, here, in this project, is debatable; at best it's research with a soft R.  CAN packet recording/replay in "a garage in San Francisco" and programming around that (while hoping not to trigger undocumented and unknown proprietary module code) isn't research, it's more akin to hackaday.io jailbreak project.

And there-in is the major issue, particularly with issues of safety.  Misleading claims such as "It's about on par with Tesla Autopilot at launch, and better than all other manufacturers" create a false sense of security with those who see the examples in action with real people.  Moreover, users, testers, and developers cannot possibly understand the entirety of the codebase and must, on some level, put trust in the environment and code maintainers/developers having an emphasis on safety.  This is impossible, however, since we're talking about a best-attempt of reverse engineering on several vehicle platforms using propriety devices and no relationships with those product manufacturers or access to complete module code being explored through CAN sniffing.

As a reminder to folks, this public repository exists because the project was deemed unsafe--in a very clear and seemingly restrained statement--by the NHTSA ("dedicated to achieving the highest standards of excellence in motor vehicle and highway safety").

And here's a choice quote from The Verge:
> Last year I met with most of the CEOs of the big auto companies. A lot of them are just hopeless, all adamantly convinced they must own self-driving technology, in house. 

There are hundreds of vehicle manufacturers the world over and Mr. Hotz met with _most_?  And they're hopeless because, in reality, they have numerous business, engineering, and IP relationships involving hundred of employees and didn't want to contract with or purchase the IP of Comma.ai?  Perhaps they know what they're looking at.

Either way, open sourcing components of this software means it moves ahead, gains traction from free press, picks up some free developers, and attempts to send a big middle-finger to the NHTSA and those who weren't interested.  Short-lived gains.

This is where there's a difference between _research_ and _Research_ and why when someone like Musk says he's unconcerned, I believe him.

research here is small group of siloed people who are attempting to stand on the shoulders of giants and claim they are now on par with major OEMs and businesses (groups that do millions of simulations, tests, safety checks, data analysis, and use a variety of redundant sensors).  Where are the relationships with the companies making the driver-assist equipment?  Where are the NDA whitepapers?  Where's the access to source code and full behavioral processes?  Where are the PhDs, engineers, refereed papers, [... cut ...] aths, and arrogance--all have their place is product design, leadership, and deployment, but even those personalities knew safety inherent in the code and for users is the first design priority.

> We believe that self driving car researchers are much safer than certain other groups of people legally allowed on public roads, namely teenagers, old people, and "buzzed" people.

Concrete suggestion: Assume they're not, get to know OEMs, work together, then program and test, test, test, accordingly.

toxicgumbo, 2016-12-12:
Despite my concerns, keep up the momentum and best wishes.  Looks like a lot of fun in the right setting and circumstances.

paramaggarwal, 2016-12-12:
(Responding, as I feel that someone needs to be vocal against the "regulate-it" argument.)

1. I appreciate your response and am able to see your perspective. It would be amazing if everything was the way you describe it.
2. I'm sure if they had 50 more people, he would assign 24 to legal, 24 to safety and continue advancing the state of technology with the rest two. The point being that just like education and health, if transportation is also going to be an ultra-regulated field - then we are going to be hurting innovation and hurting garage startups from ever being given a chance.
3. The way I understand it, they are looking for after-market and OEM manufacturers outside the US when no one in the US took interest. Also, they personally deem the AI layer as proprietary and everything else is commodity - hence the open source move. Once they get a partner and are able to work towards an integrated project, they will be able to do the kind of intensive safety-focused tests and enhancements as you mention.

I'm not completely sure what we are discussing here. If one has any ideas on how to make the code more robust and safe, they can open a PR. Also, I think his comparison to other systems was in terms of functionality/performance.

geohot, 2016-12-12:
Off topic posts will be deleted. If you have specific actionable concerns please open issues for them. This is not the correct place to promote your comma.ai competitor, invite me to come speak, discuss if certain groups have a monopoly on the word "research", or debate whether Megasquirt is the devil. Your suggestion, "Get to know OEMs" has nothing to do with openpilot, an open source driver assistance system. This is not the comma.ai forum, this is a piece of free software.

Topic locked.