commaai/openpilot #28: Complete autonomous driving?

Fryelock, 2016-12-12:
I have read in a few places that the current version of openpilot is as good/slightly better than Tesla's Autopilot 7.0, and with enough use and collaboration could compete with the new Autopilot 8.0. 

Reading the description of Enhanced Autopilot 8.0 got me thinking, how far is openpilot from full autonomy and what would it take to get there? I understand this is only available on the top of the line Tesla's. 

From https://www.tesla.com/autopilot

> Once on the freeway, your Tesla will determine which lane you need to be in and when. In addition to ensuring you reach your intended exit, Autopilot will watch for opportunities to move to a faster lane when you're caught behind slower traffic. When you reach your exit, your Tesla will depart the freeway, slow down and transition control back to you.
> 
> Your Tesla will match speed to traffic conditions, keep within a lane, automatically change lanes without requiring driver input, transition from one freeway to another, exit the freeway when your destination is near, self-park when near a parking spot and be summoned to and from your garage.

bertramt, 2016-12-15:
A new Tesla has a bunch more sensors, cameras and processing power than what is available with the NEO currently.  Also currently comma isn't doing anything with signs or stop lights which I don't think the current camera in the oneplus is good enough to see at a reasonable distance.  The more NEOs built and the more miles logged, the closer we will get to the next level. Along the way other cameras and sensors might be a requirement to make that happen.   The good news is every generation of cars is adding more sensors and better sensors to make this happen. The progress to full autonomy is progressing at an ever increasing rate.

Also This probably isn't so much a software/bug issue.  You probably should direct your inquires over to the forum. https://beta.comma.ai/

Fryelock, 2016-12-19:
Damn this is looking more like a $700 water bottle. 

BTW there is no place to direct my inquires at the link you posted, just a leaderboard.