commaai/openpilot #34: Honda Accord 2009 (EU version)

acalatrava, 2016-12-28:
I've a Honda Accord 2009 with LKAS and ACC European version. I read on the readme that Accord may need different hardware however the 2009 Accord EU is not like the US version, actually it look alike the Acura. 

Do you know if it will work with it? If not, what will I need and how can I sniff packets in order to make it compatible?

Thanks!

geohot, 2016-12-29:
Wow, didn't know they had ACC and LKAS back then. Start by sniffing the CAN bus on the ODB port and look for the packets that trigger the brakes and steering.

acalatrava, 2016-12-29:
Yes they did! Actually it was the main reason I bought this car. Ant it actually works damn well... I think there has been zero improvement in this matter during the past 6 years... unbelievable...

Ok, despite I'm a programmer and electronic hobbyist I'm a noob at car hacking... how should I start to sniff this packets? What do I need? A simple ELM327 will work? If you could point me to the right source to start with I'll be grateful.

geohot, 2016-12-29:
ELM is terrible. Build a NEO and the OBD cable in the guide. Install openpilot and it'll log the CAN packets.

hableel, 2016-12-29:
Does it matters if the car uses kmh instead of mph ? Is there something needs to be changed?

geohot, 2016-12-30:
No, CAN is largely in m/s anyway.

acalatrava, 2016-12-30:
I will probably build the neo if I have time for it, however there is a couple doubts that restrain me from doing it: 

 1) what will improve? You said that this is an improvement for the LKAS and ACC and this is not truly autonomous driving, so, what will be improved? What we'll be see? Do you have a video of this improvements?

 2) my car is manual shifting, not automatic... I understand that this could be a Strange for you since almost every car in the US is automatic shifting. When ACC+LKAS is enabled I can change shift and ACC+LKAS remain enabled. So I guess your system will only modify/improve ACC+LKAS system, so... should it work?

Thank you!!

nesherhh, 2017-01-02:
"ELM is terrible" - but, is it possible with ELM to see anything?

acalatrava, 2017-01-02:
Also, is it possible to use openpilot with the OnePlus 3 with a commercial CAN-USB interface instead of building neo board?

mslavescu, 2017-01-26:
@acalatrava which way did you go?
I got CANtact board directly from Eric, but I have a much older Accord.
Just received the DB9 to OBD II cable, I'll try to see what info I can collect and what I can control, if you find any Accord CAN related info please let me know.
For speed and throttle info you can use OBD II bluetooth adapters, works very well even on my car.

closed 2019-06-02 by pd0wm