commaai/openpilot #41: about  change lanes    overtake

leo333, 2017-01-07:
This system uses the ACC millimeter wave radar, cell phone camera, is not able to achieve automatic lane change operation and overtake?

geohot, 2017-01-07:
Can't see in the blond spot on the cars we support. If you have blind spot detection you can add quite easily.

TK300, 2017-01-08:
We need to add more cameras and sensors just like the new tesla cars. Especially a long ranging detection to the left rear to detect fast passing cars. I got a car from 1986 and i might be able to add the hardware to it, i just dont know much about programming though. Would it be easier to transfer the things from an existing car or buy the parts seperately and make it work with a raspberry pi or an arduino maybe?

leo333, 2017-01-13:
how do you think NVIDIA Jetson TX1  , $430
Video decode	4K 60 Hz
Video encode	4K 30 Hz
CSI	Up to 6 cameras | 1400 Mpix/s
Display	2x DSI, 1x eDP 1.4, 1x DP 1.2/HDMI
Connectivity	Connects to 802.11ac Wi-Fi and Bluetooth-enabled devices

juanlvo, 2018-01-30:
For changing lines, we need more cameras and Lidiar to detect cars, pedestrians and bicycles in the others line, and the fusion all the data to be 100% sure we can execute a changing of lines.

CRV-Driver, 2018-04-07:
Mine changes lanes all the time!  I wish it wouldn't. It needs more work on lane detection. It has a hard time with yellow lines, yellow dashed lines, glare, slanted roads, and when it loses one of the lines it hugs the other, and eventually drives right over it.  Please help! (it's been calibrated on several different kinds of roads)

closed 2019-06-02 by pd0wm