commaai/openpilot #42: adb: error:  Read-only file system

droter, 2017-01-10:
Hi,

I am getting this error when trying to run the ./install.sh script:
adb: error: failed to copy 'files/id_rsa_openpilot_ro' to '/tmp/id_rsa_openpilot_ro': Read-only file system

Model Number:  ONEPLUS A3000
Android Version:  6.0.1
Android security patch level: November 1, 2016
Build number:  ONEPLUS A3000_16_161118

Any ideas?

Thanks,

Matt

geohot, 2017-01-10:
You have to install NEOS first. openpilot doesn't run on Android.

fbagnol, 2017-01-21:
Hi, 

Sorry but I don't find how to install NEOS, can you give a link ?

FredB

jleonar, 2017-01-21:
Have you built the board?  https://github.com/commaai/neo

The instructions are all there.

fbagnol, 2017-01-22:
OK, found!

It's a zip archive on the release section.

closed 2017-02-23 by droter