commaai/openpilot #44: Transfer to BMW-Cars

kaeferfreund, 2017-01-12:
Hey George,

I love to see what you are doing here and I'd wish to try the system. But at my place we only got BMW's. These also have radar-sensors and adaptive cruise control.

What is needed to add a BMW Port to open pilot?
I could do a can-log/-dump for you on F10, F20 (LCI) and other models, if that helps...

geohot, 2017-01-14:
It would take a bit of engineering. It's very doable by an outside party given what we've released.

derek99401, 2018-03-01:
I got a F32, the car's can msg from ODBII port (using Panda) only gives heartbeat message with msg ID 0:130. Some research online shows that you have to tap into other port in order to get more useful data stream.

triscb82, 2018-08-25:
I have a fully loaded G30 along with all the technical documentation.  I am looking for someone who has programming knowledge to help me get OpenPilot running on it.  Who’s game??

greutair, 2018-09-13:
Would it be working with the BMW i3 not having built in a radar?

jfrux, 2018-09-13:
You can lookup the current “status” of any given vehicle year, make, model, trim at https://opc.ai/lookup

closed 2019-06-02 by pd0wm

MrQwertyDude07, 2021-05-18:
Hello! Would this work on a fully loaded 2021 BMW Z4

kaeferfreund, 2021-08-19:
I found the latest BMW ADAS with iStep 03/2021 to be very close to open pilot in regards of lane keeping + there are more functions e.g. reading the speed signs and using that as the acc set speed

@geohot try the latest BMW ADAS (I know, its mobileye) but since the 03/2021 OTA update (yes they update the whole car!) it really made OP a lot less attractive :/

triscb82, 2022-01-06:
> Hi, I just heard about this. I'm wanting to know if BMW i3 support is
> possible. It just feels like a perfect candidate for this, a vehicle with
> early generation ACC and lane keep that BMW just... Abandoned.
>
> I'm amazed this project even exists, but honestly, surprised it doesn't
> already support the i3 as that's something I would've thought of quickly
> alongside the Leaf and Bolt when thinking of "the poor man's Tesla," and
> evidently it has the sensors and hardware to turn the steering column, so
> it seems like a good candidate.
>
> First question: How come this issue is closed? That would indicate support
> was added, and I do see some merged pulls, but I don't see any BMWs on the
> support list.
>
> Second question: If the i3/other BMWs are not supported, what would it
> take to add them? I have a cheap ODBII adapter, used CANBUS once to make
> two Arduinos talk to each other, know C++ and Python very well, and have a
> little AI dev experience too, but uh, I know nothing about vehicles at all,
> other than that the i3 is basically a giant RC toy car with lots of torque
> and I love it. @derek99401 <https://github.com/derek99401> mentioned you
> need to tap into more ports than ODBII?

This got abandoned due to significant issues supporting FlexRay.  The
industry has abandoned the protocol but getting the DBC’s for the vehicles
has proven to be near impossible.  The key guy I had working on the project
fell off the face of the earth and we ran out of funding and resources.  I
honestly don’t see any flexray vehicle being supported.  I was super
optimistic and excited but have utterly lost hope due to there being so
many variables in reliably being able to talk on the flexray bus.

Pecacheu, 2022-01-07:
Ah, I see. I honestly didn't know that all modern cars don't use CANBUS. So I take it from Wikipedia, FlexRay is an alternative protocol that some vehicles used to use, including BMWs. I was able to find [this rather relevant thread](https://www.mybmwi3.com/forum/viewtopic.php?t=16472) just form a search for 'FlexRay BMW', actually.

And yeah, that sucks. Something tells me you can't just go out and buy an Arduino shield for this like you can with CANBUS (in fact I already have one, as I mentioned). I couldn't find any shields in a quick search, though I did find mention of a a FlexRay library, and [whatever the heck this is](https://www.aliexpress.com/item/32912039329.html).

Pecacheu, 2023-11-01:
If anyone has any info on FlexRay development, I could give a crack at it. I don't think an Arduino has enough CPU power to handle the throughput, but a Raspberry Pi surely should. But I can find little info about the protocol or adapters available. And without the ability to use software like SAVVYCAN, reverse engineering might be difficult. Honestly, I almost just want to rip the car apart and hook wires straight up to the power steering motor.

On the plus side, folks in the i3 development Discord have figured out the steering angle sensor and some other sensors available on the CANBUS lines. I have given this a test in SAVVYCAN and can confirm it works. Power steering control is definitely on the FlexRay bus though, not CAN.