commaai/openpilot #45: Calibration

TheMutley, 2017-01-16:
you could end up with a blank calibration file ( for example, if you shutdown the phone while the calibration deamon is writing the file) and the calibration process seem to stop doing it's job properly after this. i noticed that i had a blank calibration file in /sdcard that dated from january 1st even tough i did a lot of driving since then. the file was there , but zero byte in size. i deleted the file and the whole calibration process restarted. I just did an hour of driving and it's now perfectly aligned.

mgraczyk, 2017-01-16:
I'm working on a simple fix in which we never intentionally write an invalid calibration file and discard an existing file if it is invalid. This should be in the next openpilot release.

mgraczyk, 2017-02-01:
Fixed in 0.2.4

closed 2017-02-01 by mgraczyk