commaai/openpilot #48: Volkswagen Polo XL from 2003

aifrs96, 2017-01-26:
I have a VW Polo XL from 2003.
Would this be possible in this car?

Thanks in advance!

bertramt, 2017-02-02:
Without doing any research my guess would be no based on the age of the car.  

Things your car would need to be fully supported.
- Electronic power steering.  This was not common in older cars.
- Front radar sensor.    I'm not sure any cars offered this in 2003
- LKAS (Lane Keeping Assist System).  Again I'm not sure any cars offered this in 2003
- A way to electronically apply the brakes.  Unless your car has some sort of frontal radar odds are you don't have this.

People are looking for ways to add this tech to older vehicles but don't expect it to be easy.  I can't comment for comma, but their focus is largely on newer vehicles where the system is much more plug and play.  You would largely be on your own to make something like this work on your vehicle.

mapsonburt, 2017-04-07:
I have a 1908 Model T, would it work with mine?    :-)

legonigel, 2019-03-13:
Closing this. Any further discussion can continue on [discord](http://discord.comma.ai) on #other-vehicles or #topic-without-lkas

closed 2019-03-13 by legonigel