commaai/openpilot #50: ACC on Non-Sensing 2016 Civic EX-L

Krishtof-Korda, 2017-02-04:
I just bought the 2016 Civic EX-L, **without** Honda Sensing package. I think it would be really cool to add ACC to this car for my wife. Are there enough messages on the CAN bus to make this happen or have they been disabled on non-sensing vehicles. I really only need an 'apply brake = 0.xx' and 'apply accelerator = 0.xx' message to accomplish what I am trying to do. I am not familiar yet with the CAN bus. What would I need to start sniffing the CAN and see if the appropriate messages are available?

If possible, I plan to build the kit and do some testing. I just wanted to ask those who have experience with this before dropping the coin to build the Comma One. 

Thanks for any help anyone can provide. I really think this would be a fun project and a way I could contribute to expanding capabilities for openpilot on non-sensing cars.

geohot, 2017-02-04:
Here's the problem:

https://www.hondapartsnow.com/genuine/honda~motor~eps~service~53651-tba-a21.html is the steering controller from a non Honda Sensing trim.
https://www.hondapartsnow.com/genuine/honda~motor~eps~service~53651-tba-a31.html is the steering controller from the same trim with sensing.

I doubt the non sensing controller accepts the apply steer torque command. Possible to fix with a reflash maybe, but this could be a rabbit hole. You are also missing a radar, which eventually when vision gets better may be a non issue, but for now will provide a low quality experience.

Krishtof-Korda, 2017-02-04:
If I am not looking to steer the vehicle but just do accelerator and brake, do you think that could work with the OnePlus3 vision only at least for experimenting and testing? Then later maybe I could add the radar.

Also do you have a link for good resources on were to get smart on CAN bus and sniffing?

Thanks again for any help.

geohot, 2017-02-04:
Same problem as with the steering for brakes, and actually it's probably even a hardware change this time and worse.

http://parts.sonshonda.com/showAssembly.aspx?ukey_assembly=1633703&ukey_make=988&ukey_model=13654&modelYear=2016&ukey_driveLine=5541&searchString=vsa

leo333, 2017-02-22:
radar
https://www.hondapartsnow.com/parts-list/2016-honda-civic-4dr-tour-ka-cvt/camera-radar.html?PNC=3

leo333, 2017-05-22:
@geohot 
How do you start the ACC module in the car after the radar is added?

closed 2019-06-02 by pd0wm