commaai/openpilot #53: OpenPilot name is not completely defunct... please respect it

MAVProxyUser, 2017-02-22:
Some of us are still around and even *selling* hardware using the OpenPilot name, thanks! 

OpenPilot ground / flight controller: 
https://sasquatchlabs.org/product/genuine-nano-flight-controller-w-case
OpenPilot 433mhz telemetry radio aka OPLink:
https://sasquatchlabs.org/product/openpilot-oplink-mini-modem-bundle

MAVProxyUser, 2017-02-22:
The name is also Trademarked... 
https://search.ipaustralia.gov.au/trademarks/search/view/1362707?q=openpilot

geohot, 2017-02-22:
According to the docs, http://opwiki.readthedocs.io/en/latest/user_manual/gcs_install_op.html I quote (sorry about the caps), "OPENPILOT WAS DISCONTINUED!". 

openpilot.org is also down as of April last year.

MAVProxyUser, 2017-02-22:
What part of those are NOT the docs do you fail to comprehend? If I put up a random web page and say that Comma AI was discontinued does that make it so? Of course not!?

MAVProxyUser, 2017-02-22:
Likewise that has NOTHING to do with the Trademark...

MAVProxyUser, 2017-02-22:
Yes... the web server has been pulled, that does not give you the right to acquire the name... Nor does it make is classy. 

https://github.com/openpilot/OpenPilot

You can see here the official project Wiki *was* at:
Wiki, docs and manuals:	http://wiki.openpilot.org/

NOT The random readthedocs.io website you furnished...

MAVProxyUser, 2017-02-22:
I hear your project is defunct... did you read the docs? 
https://readthedocs.org/projects/openpilot-by-comma-ai/

geohot, 2017-02-22:
[image]

MAVProxyUser, 2017-02-22:
Actually Geo... I used to respect you and have helped you in a former life... right now you are simple being flat out disrespectful and hovering over the corpse of another successful autopilot name.. lets be honest.

geohot, 2017-02-22:
That github you linked hasn't been updated since 2015. That link you posted is dead. And you filed 3 issues, not a very respectful way to address something.

MAVProxyUser, 2017-02-22:
I saw how you blew the first guy off... clearly you had no respect why should I come in doing anything other than calling a spade a spade? 

Again... keep acting like you don't know it is trademarked... project status has NOTHING to do with the name being trademarked and you know it. 

I mean come on you keep citing some BS documentation as if you knew ANYTHING about the original project.

MAVProxyUser, 2017-02-22:
Hey guys... I have this new project called Blackra1n (no one uses that project any more so I can just snag the name!)... you should come check it out! *roll eyes*

closed 2017-02-22 by geohot