commaai/openpilot #54: OpenPilot name is trademarked in Australia FWIW...

MAVProxyUser, 2017-02-22:
https://search.ipaustralia.gov.au/trademarks/search/view/1362707?q=openpilot

Goods & Services
Class 9: Auto-pilots; computer software

27-JAN-2011	Trade Marks Registered	Advert
10-JAN-2011	Register Trade Mark	Update
10-JAN-2011	Amend Status from Accepted	Update
28-SEP-2010	Registration fee (1 class)	Correspondence
23-SEP-2010	Accepted Applications for Registration of Trade Mark	Advert
23-SEP-2010	Accepted Applications for Registration of Trade Mark	Advert
03-SEP-2010	Amend Status from Awaiting Advertisement	Update
15-AUG-2010	Clear Report - Approved	Report No. 1
15-AUG-2010	Clear Examination report approved	Update
10-JUN-2010	Applications Filed	Advert
25-MAY-2010	Amend Status from Indexed	Update
25-MAY-2010	Amend Status from Filed - Approved	Update
21-MAY-2010	Approve Filing	Update
20-MAY-2010	Filing Elec Appl'n for TM - Pick List 1 Class	Correspondence

geohot, 2017-02-22:
Duplicate issue.

closed 2017-02-22 by geohot

MAVProxyUser, 2017-02-22:
This isn't a duplicate issue.... you have not addressed trademark concerns.