commaai/openpilot #57: Minimum requirements for vehicle

abidbodal, 2017-02-22:
Is there a list of minimum requirements for a car to be retrofitted to use openpilot sw? I can figure out the CAN packets and do the parsing. I have a 2007 Toyota Prius that I am considering trying this on.

How is the Radar sensor data used and can the system work without it or are there aftermarket radar sensor s that can be installed?

geohot, 2017-02-22:
For radar, you can buy a radar from a junkyard car off ebay and install it.

afeezaziz, 2017-03-02:
Hi,

I am wondering whether can use the openpilot sw to actually send data back to control the car? My car is 2012 Toyota Prius C(Asian Model).  Is there a list of cars that can actually do this? Or all North American cars can do but perhaps not Asian cars?

eleben, 2017-05-18:
Hello,

@bidbodal - did you every do anything with the Prius? I'm looking at getting either one with radar and lane keep assist or a Ford Fusion Energi and trying it out once figured out

abidbodal, 2017-05-19:
No I didn't unfortunately because I needed to install radar sensors on my
2007 Prius

closed 2019-06-02 by pd0wm