commaai/openpilot #64: i can't train "train_generation_model"successfully..please help

zhaohuaqing1993, 2017-02-27:
when i train"train_generation_model",i  meet the problem
Traceback (most recent call last):
  File "./train_generative_model.py", line 160, in <module>
    g_train, d_train, sampler, saver, loader, extras = get_model(sess=sess, name=args.name, batch_size=args.batch, gpu=args.gpu)
  File "/home/sues/Desktop/research-master/research-master/models/autoencoder.py", line 171, in get_model
    d_loss_legit = tf.reduce_mean(tf.nn.sigmoid_cross_entropy_with_logits(D_legit, tf.ones_like(D_legit)))
  File "/home/sues/anaconda2/lib/python2.7/site-packages/tensorflow/python/ops/nn_impl.py", line 146, in sigmoid_cross_entropy_with_logits
    _sentinel, labels, logits)
  File "/home/sues/anaconda2/lib/python2.7/site-packages/tensorflow/python/ops/nn_ops.py", line 1533, in _ensure_xent_args
    "named arguments (labels=..., logits=..., ...)" % name)
ValueError: Only call `sigmoid_cross_entropy_with_logits` with named arguments (labels=..., logits=..., ...)
if someone do it successfully?
thanks

espes, 2017-02-28:
Not an openpilot issue

closed 2017-02-28 by espes

zhaohuaqing1993, 2017-02-28:
I am so sorry that I don't understand crealy, should I build a platform of openpilot?could you tell me in detail thank you very much