commaai/openpilot #68: Hyundai Ioniq? Already has ACC & LKAS

phr00t, 2017-03-02:
If I were to get a Hyundai Ioniq, is it possible to use this software with it, considering it already has these features? I see something about the CAN-bus hacking... wondering if anyone has ever tried this on a Hyundai or know how to get started.

mniami, 2017-10-02:
I will love to use test it with my Hyundai i30, let us know is it feasible.

lambaek, 2017-11-12:
Same goes for me. Would love to test on hyundai i30

cori2000, 2017-11-12:
I would test it for my Hyundai i40. Could you share some more info. on it?

VincentVonPark, 2018-10-26:
is it possible to test with Hyundai Accent Manual?

bugsy924, 2018-10-26:
You can only control a car that already has built in safety features

Built in LKAS means you can add steering
Built in ACC means you can add accelerator+brake

emmertex, 2019-01-15:
IONIQ shoudln't be any issue.
as for i30 and i40, only if they have stock LKAS

If so, safe to assume they will work fine.
join hyundai-kia-genesis on Discord

closed 2019-09-11 by rbiasini

Growiel, 2020-01-26:
I see that this is closed, is the Ioniq now supported ? I'm getting an Electric Ioniq soon and am interrested in Openpilot. Thanks !

soulslicer, 2021-02-07:
any luck? its been a year?