commaai/openpilot #70: Add 2017 Honda CR-V with Sensing

energee, 2017-03-05:
I wanted to get some conversation started on this car.

The 2017 has blind spot detection which could be a useful for lane changing in the future. Plus there isn't an SUV option yet for openpilot.

Checked things out over the CAN. Most of the ID's match up with the Civic ID's in fingerprint.py. 

Didn't see the 0x405 ID for open doors coming through cansniffer. This message is sent when a door is open?

The "USER_BRAKE" id matches up, but didn't see anything for "BRAKE_PRESSED" on 0x17c, is this sent when brake is being pressed by ACC?

Going to check messaging out with ACC and LKAS on tomorrow. What signals should I be looking for?

geohot, 2017-03-08:
I believe the brake is the same but steering changed. If I dig around, I looked into the steering on this car, it was almost the same packet.

This is probably the easiest car to port to.

energee, 2017-03-19:
The CR-V doesn't use the same JAE Connector for the LKAS Camera as the civic, it looks like a 12 pin, picture attached.
[image]

geohot, 2017-03-20:
Oh, you'll find the pinouts with a subscription to techinfo.honda.com

energee, 2017-03-20:
Cool, image attached.
[image]

energee, 2017-03-20:
Could not find a male pin connector on digikey, will contact their support with pitch and row spacing.

zhaohuaqing1993, 2017-03-20:
@geohot Hi，i run the ./view_generative_model.py transition --name transition and always see the string "loading weights...",what should i do?thanks

geohot, 2017-03-20:
Hmm, seems to lack an always on power rail. But otherwise, it's the same 2 CANs.

zhaohuaqing1993, 2017-03-20:
I just want to see the model's gif,I follow what you say in "research ",but always say "loading weights "

energee, 2017-03-20:
Can IG1 wire be used for power?

Is it possible for the NEO to issue a "power on" and "off" command based on external power state?

djjohnny1093, 2017-03-20:
Guys, any shot of me being able to get this to work on my 2017 honda crv touring. I gotta drive to florida mid april, i have minimal programming knowledge. Can anyone create a beginning to end manual with workarounds and instructions that need to be done for the crv touring? Because its not plug and play like it is with the civic? If someone can help me ill make a youtube video. There is literally nothing on youtube. Maybe im looking for the wrong thing? Sorry if i sound stupid, i wanna help but dont know how.

energee, 2017-03-20:
@djjohnny1093 Any collaboration on the CR-V will probably take place here. Should be plug and play once complete.

geohot, 2017-03-21:
Interesting. The 2016 CR-V Touring matches the Civic with connector. It changed on the 2017, so it'll be a bit different.

zhaohuaqing1993, 2017-03-21:
@geohot  hi could you tell me how to solve this promble?https://github.com/commaai/research/issues/40

djjohnny1093, 2017-03-21:
I live in new york...anybody over here know how to do this stuff and wanna tinker with my car so we can figure this out?

djjohnny1093, 2017-03-22:
So @energee, your working on this currently? Are you developing the technique for us crv owners and then we will just follow your lead? If so thats amazing and extremely appreciated. Thankyou.

energee, 2017-03-29:
DBC File added, there are some notes added on missing messages

energee, 2017-03-30:
`9.999999   94  80 1C C8 2A 04 00 0C 51 ...*...Q
0.760001   C7  00 18 D8 16             ....
9.999999   E7  14 05 40 A0 38          ..@.8
9.999999  130  00 00 FF FF 00 00 04 31 .......1
9.999999  14A  02 E5 00 00 07 02 E5 08 ........
9.999999  158  00 00 00 00 00 00 08 11 ........
9.999999  17C  00 00 03 5A 00 00 00 3F ...Z...?
9.999999  191  01 00 00 A9 A9 21 01 11 .....!..
9.999999  1A4  00 66 00 00 00 00 00 0D .f......
9.999999  1A7  00 06                   ..
9.999999  1AB  80 02 35                ..5
9.999999  1AC  7F FF 00 00 00 00 00 0D ........
9.999999  1B0  00 00 00 00 00 01 47    ......G
9.999999  1C2  00 00 00 00 00 00 00 09 ........
9.999999  1D0  00 00 00 00 00 00 00 0A ........
9.999999  1D5  46 E5 35                F.5
9.999999  1D6  00 13                   ..
9.999999  1DA  19 00 FA 00 00 00 00 1C ........
9.999999  1DC  02 03 5A 00 00 00 19    ..Z....
9.999999  1DF  8A D0 01 00 00 00 00 1A ........
9.999999  1ED  00 00 00 00 1B          .....
9.999999  1EF  03 FF 00 75 30 00 00 19 ...u0...
9.999999  1FB  0D                      .
9.999999  221  1E 00 80 00 30 18       ....0.
9.999999  255  00 00 00 00 55 55 68 EE ....UUh.
9.999999  295  00 00 00 17             ....
9.999999  296  00 00 00 16             ....
9.999999  305  80 54 00 00 00 00 0F    .T.....
9.999999  309  00 00 00 00 00 00 81 8B ........
9.999999  30C  00 00 00 FF 00 40 C0 0B .....@..
9.999999  31B  00 00 00 00 00 00 00 09 ........
9.999999  320  00 00 00 00 00 00 00 30 .......0
9.999999  324  61 38 0A 08 00 00 00 38 a8.....8
9.999999  326  0A BE 00 00 00 00 00 0A ........
9.999999  328  00 00 00 00 00 80 00 7C .......|
9.999999  32E  84 00 02 25             ...%
9.999999  32F  00 80 00 00 00 00 00 39 .......9
9.999999  331  3D 00 00 01             =...
9.999999  339  FE 1A 50 2A             ..P*
9.999999  33D  00 00 10 50 0F          ...P.
9.999999  35E  00 00 00 00 00 00 00 02 ........
9.999999  371  00 00 00 00 00 00 5E 82 ......^.
9.999999  372  E0 10 00 0D             ....
9.999999  374  32 60 01 00 00 40 40 06 2`...@@.
9.999999  378  00 00 00 00 06 40 00 48 .....@.H
9.999999  37B  50 00 00 20 18 00 00 03 P.. ....
9.999999  396  00 00 31 32 01 00 0C    ..12...
9.999999  39F  00 00 00 00 00 00 00 0D ........
9.999999  3A1  7C 00 10 02 00 00 01 03 |.......
9.999999  3D7  B0 C1 04 00 00 00 00 32 .......2
9.999999  403  00 00 00 00 3E          ....>
9.999999  405  00 00 00 00 00 00 00 1E ........
9.999999  40C  54 4E 17 0F 03 01 00 11 TN......
9.999999  40F  52 07 4A 07 4E 07 4E 34 R.J.N.N4
9.999999  428  00 00 00 00 09 FB 16    .......
9.999999  444  1B                      .
9.999999  454  4D FA 01 3A 63 00 60 13 M..:c.`.
9.999999  465  00 00 00 00 00 00 01 17 ........
9.999999  467  20 14                    .
9.999999  510  00 00 00 00 00 52 00 38 .....R.8
9.999999  516  00 09 FB 00 00 00 00 36 .......6
`
Full Dump of messages (Car is not moving)

energee, 2017-04-07:
Dump of messages from the ACC Can line at the camera:

```
0.000000   E4  00 00 00 00 33          ....3
0.000000   E5  04 00 80 10 00 00 00 35 .......5
0.000000  201  10 AC 00 28 A9 00 00 2F ...(.../
0.000000  240  80 0A 80 00 00 00 00 26 .......&
0.000000  241  A0 00 7F F0 7F F0 80 23 .......#
0.000000  243  80 0A 80 00 00 00 00 23 .......#
0.000000  244  A0 00 7F F0 7F F0 80 20 ....... 
0.000000  246  80 0A 80 00 00 00 00 20 ....... 
0.000000  247  A0 00 7F F0 7F F0 80 2D .......-
0.000000  249  80 09 80 00 00 00 00 1F ........
0.000000  24A  90 00 7F F0 7F F0 80 1C ........
0.000000  280  FE F0 FF F9 FF FE FF 12 ........
0.000000  281  FC FF FF F2 FF FF 00 18 ........
0.000000  282  FF F9 FF C0 FF FF FF 99 ........
0.000000  283  80 12 80 00 80 00 FF 11 ........
0.000000  284  FE F0 FF F9 FF FE FF 1E ........
0.000000  285  FC FF FF F2 FF FF 00 14 ........
0.000000  286  FF F9 FF C0 FF FF FF 95 ........
0.000000  287  80 12 80 00 80 00 FF 1D ........
0.000000  288  FE F0 FF F9 FF FE FF 1A ........
0.000000  289  FC FF FF F2 FF FF 00 10 ........
0.000000  28A  FF F9 FF C0 FF FF FF 91 ........
0.000000  28B  80 12 80 00 80 00 FF 19 ........
0.000000  28C  FE F0 FF F9 FF FE FF 16 ........
0.000000  28D  FC FF FF F2 FF FF 00 1C ........
0.000000  28E  FF F9 FF C0 FF FF FF 9D ........
0.000000  28F  80 12 80 00 80 00 FF 15 ........
0.000000  290  FF D2 FF 80 80 00 7D 1D ......}.
0.000000  291  FF D2 FF 80 80 00 7D 1C ......}.
0.000000  292  FF D2 FF 80 80 00 7D 1B ......}.
0.000000  293  FF D2 FF 80 80 00 7D 1A ......}.
0.000000  294  FF D2 FF 80 80 00 7D 19 ......}.
0.000000  295  FF D2 FF 80 80 00 7D 18 ......}.
0.000000  296  FF D2 FF 80 80 00 7D 17 ......}.
0.000000  297  FF D2 FF 80 80 00 7D 16 ......}.
0.000000  2C8  FF D2 FF 80 80 00 7D 12 ......}.
0.000000  2C9  FF D2 FF 80 80 00 7D 11 ......}.
0.000000  2CA  FF D2 FF 80 80 00 7D 10 ......}.
0.000000  2CB  FF D2 FF 80 80 00 7D 1F ......}.
0.000000  2CC  FF D2 FF 80 80 00 7D 1E ......}.
0.000000  2CD  FF D2 FF 80 80 00 7D 1D ......}.
0.000000  2CE  FF D2 FF 80 80 00 7D 1C ......}.
0.000000  2CF  FF D2 FF 80 80 00 7D 1B ......}.
0.000000  2D0  FE F0 FF F9 FF FE FF 1D ........
0.000000  2D1  FC FF FF F2 FF FF 00 13 ........
0.000000  2D2  FF F9 FF C0 FF FF FF 94 ........
0.000000  2D3  80 12 80 00 80 00 FF 1C ........
0.000000  2D4  FE F0 FF F9 FF FE FF 19 ........
0.000000  2D5  FC FF FF F2 FF FF 00 1F ........
0.000000  2D6  FF F9 FF C0 FF FF FF 90 ........
0.000000  2D7  80 12 80 00 80 00 FF 18 ........
0.000000  2D8  FE F0 FF F9 FF FE FF 15 ........
0.000000  2D9  FC FF FF F2 FF FF 00 1B ........
0.000000  2DA  FF F9 FF C0 FF FF FF 9C ........
0.000000  2DB  80 12 80 00 80 00 FF 14 ........
0.000000  2DC  FE F0 FF F9 FF FE FF 11 ........
0.000000  2DD  FC FF FF F2 FF FF 00 17 ........
0.000000  2DE  FF F9 FF C0 FF FF FF 98 ........
0.000000  2DF  80 12 80 00 80 00 FF 10 ........
0.000000  2E0  FE F0 FF F9 FF FE FF 1C ........
0.000000  2E1  FC FF FF F2 FF FF 00 12 ........
0.000000  2E2  FF F9 FF C0 FF FF FF 93 ........
0.000000  2E3  80 12 80 00 80 00 FF 1B ........
0.000000  2E4  FE F0 FF F9 FF FE FF 18 ........
0.000000  2E5  FC FF FF F2 FF FF 00 1E ........
0.000000  2E6  FF F9 FF C0 FF FF FF 9F ........
0.000000  2E7  80 12 80 00 80 00 FF 17 ........
0.000000  2E8  FE F0 FF F9 FF FE FF 14 ........
0.000000  2E9  FC FF FF F2 FF FF 00 1A ........
0.000000  2EA  FF F9 FF C0 FF FF FF 9B ........
0.000000  2EB  80 12 80 00 80 00 FF 13 ........
0.000000  2EC  FE F0 FF F9 FF FE FF 10 ........
0.000000  2ED  FC FF FF F2 FF FF 00 16 ........
0.000000  2EE  FF F9 FF C0 FF FF FF 97 ........
0.000000  2EF  80 12 80 00 80 00 FF 1F ........
0.000000  2F0  FE F0 FF F9 FF FE FF 1B ........
0.000000  2F1  FC FF FF F2 FF FF 00 11 ........
0.000000  2F2  FF F9 FF C0 FF FF FF 92 ........
0.000000  2F3  80 12 80 00 80 00 FF 1A ........
0.000000  2F4  FE F0 FF F9 FF FE FF 17 ........
0.000000  2F5  FC FF FF F2 FF FF 00 1D ........
0.000000  2F6  FF F9 FF C0 FF FF FF 9E ........
0.000000  2F7  80 12 80 00 80 00 FF 16 ........
0.000000  2F8  FE F0 FF F9 FF FE FF 13 ........
0.000000  2F9  FC FF FF F2 FF FF 00 19 ........
0.000000  2FA  FF F9 FF C0 FF FF FF 9A ........
0.000000  2FB  80 12 80 00 80 00 FF 12 ........
0.000000  2FC  FE F0 FF F9 FF FE FF 1F ........
0.000000  2FD  FC FF FF F2 FF FF 00 15 ........
0.000000  2FE  FF F9 FF C0 FF FF FF 96 ........
0.000000  2FF  80 12 80 00 80 00 FF 1E ........
0.000000  33D  00 00 00 08 3A          ....:
0.000000  400  00 80 00 39             ...9
0.000000  410  54 4C 41 41 30 0D       TLAA0.
0.000000  420  00 00 00 00 00 00 00 3F .......?
0.000000  669  00 02 74 7A 68 8E 00 01 ..tzh...
0.000000  750  00 8D 00 17 00 00 00 78 .......x

```

energee, 2017-04-07:
Dump of messages from the F-Can Line at the camera:
```
0.000000   39  00 00 1B                ...
0.000000   94  80 1D 47 D6 03 00 0C 62 ..G....b
0.000000   E4  00 00 00 00 33          ....3
0.000000   E5  04 00 80 10 00 00 00 35 .......5
0.000000   E8  00 01 00 00 7E 00 1B    ....~..
0.000000  130  00 00 FF FF 00 00 04 B9 ........
0.000000  14A  FF B0 00 00 07 FF B0 3D .......=
0.000000  154  00 00 1D 00 00 00 00 3D .......=
0.000000  158  00 00 00 00 00 00 00 37 .......7
0.000000  17C  00 00 03 76 01 00 20 3E ...v.. >
0.000000  18F  00 3C 00 00 70 01 36    .<..p.6
0.000000  191  81 00 40 A9 A9 20 01 34 ..@.. .4
0.000000  1A4  01 0C 0C 64 10 00 01 04 ...d....
0.000000  1AB  80 00 37                ..7
0.000000  1B0  00 0A 00 00 00 00 4E    ......N
0.000000  1B9  00 00 00 00 03          .....
0.000000  1BE  F6 00 27                ..'
0.000000  1C2  23 00 00 00 00 00 00 04 #.......
0.000000  1D0  00 00 00 00 00 00 00 0A ........
0.000000  1DA  19 00 FA 0C 00 00 00 10 ........
0.000000  1DD  00 21 04 C8 00 00 00 11 .!......
0.000000  1DF  8A D0 01 00 00 00 00 1A ........
0.000000  1EA  00 00 00 00 00 00 04 0B ........
0.000000  1EF  03 FF 00 75 30 00 00 19 ...u0...
0.000000  201  10 AC 00 28 A9 00 00 2F ...(.../
0.000000  221  1F 00 80 00 30 17       ....0.
0.000000  240  80 0A 80 00 00 00 00 26 .......&
0.000000  241  A0 00 7F F0 7F F0 80 23 .......#
0.000000  243  80 0A 80 00 00 00 00 23 .......#
0.000000  244  A0 00 7F F0 7F F0 80 20 ....... 
0.000000  246  80 0A 80 00 00 00 00 20 ....... 
0.000000  247  A0 00 7F F0 7F F0 80 2D .......-
0.000000  249  80 09 80 00 00 00 00 1F ........
0.000000  24A  90 00 7F F0 7F F0 80 1C ........
0.000000  255  00 00 00 00 55 55 9C BA ....UU..
0.000000  280  FE F0 FF F9 FF FE FF 12 ........
0.000000  281  FC FF FF F2 FF FF 00 18 ........
0.000000  282  FF F9 FF C0 FF FF FF 99 ........
0.000000  283  80 12 80 00 80 00 FF 11 ........
0.000000  284  FE F0 FF F9 FF FE FF 1E ........
0.000000  285  FC FF FF F2 FF FF 00 14 ........
0.000000  286  FF F9 FF C0 FF FF FF 95 ........
0.000000  287  80 12 80 00 80 00 FF 1D ........
0.000000  288  FE F0 FF F9 FF FE FF 1A ........
0.000000  289  FC FF FF F2 FF FF 00 10 ........
0.000000  28A  FF F9 FF C0 FF FF FF 91 ........
0.000000  28B  80 12 80 00 80 00 FF 19 ........
0.000000  28C  FE F0 FF F9 FF FE FF 16 ........
0.000000  28D  FC FF FF F2 FF FF 00 1C ........
0.000000  28E  FF F9 FF C0 FF FF FF 9D ........
0.000000  28F  80 12 80 00 80 00 FF 15 ........
0.000000  290  FF D2 FF 80 80 00 7D 1D ......}.
0.000000  291  FF D2 FF 80 80 00 7D 1C ......}.
0.000000  292  FF D2 FF 80 80 00 7D 1B ......}.
0.000000  293  FF D2 FF 80 80 00 7D 1A ......}.
0.000000  294  FF D2 FF 80 80 00 7D 19 ......}.
0.000000  295  FF D2 FF 80 80 00 7D 18 ......}.
0.000000  296  00 00 00 16             ....
0.000000  297  FF D2 FF 80 80 00 7D 16 ......}.
0.000000  2C8  FF D2 FF 80 80 00 7D 12 ......}.
0.000000  2C9  FF D2 FF 80 80 00 7D 11 ......}.
0.000000  2CA  FF D2 FF 80 80 00 7D 10 ......}.
0.000000  2CB  FF D2 FF 80 80 00 7D 1F ......}.
0.000000  2CC  FF D2 FF 80 80 00 7D 1E ......}.
0.000000  2CD  FF D2 FF 80 80 00 7D 1D ......}.
0.000000  2CE  FF D2 FF 80 80 00 7D 1C ......}.
0.000000  2CF  FF D2 FF 80 80 00 7D 1B ......}.
0.000000  2D0  FE F0 FF F9 FF FE FF 1D ........
[... 59 lines cut ...]
0.000000  39F  00 20 00 00 00 00 00 29 . .....)
0.000000  3A1  00 C0 00 02 00 00 01 38 .......8
0.000000  3D9  00 80 34                ..4
0.000000  400  00 00 00 00 31          ....1
0.000000  403  00 00 00 00 2F          ..../
0.000000  405  00 00 00 00 00 00 00 1E ........
0.000000  40C  A1 32 48 4B 52 57 32 23 .2HKRW2#
0.000000  40F  B1 0E 17 01 04 00 00 2C .......,
0.000000  410  54 4C 41 41 30 0D       TLAA0.
0.000000  420  00 00 00 00 00 00 00 3F .......?
0.000000  454  46 F4 01 34 CE 00 60 24 F..4..`$
0.000000  516  00 0C 33 00 00 00 00 19 ..3.....
0.000000  52A  00 00 00 06 2F          ..../
0.000000  551  00 00 00 06 16          .....
0.000000  555  00 00 00 04 05          .....
0.000000  590  00 00 00 06 13          .....
0.000000  640  C4 16 68 06 03          ..h..
0.000000  641  13 FF F9 00 00 00 03 00 ........
0.000000  652  00 00 00 06 05          .....
0.000000  661  20 B0 21 61 90 00 00 29  .!a...)
0.000000  669  00 02 74 7A 68 8E 00 01 ..tzh...
0.000000  686  C0 29 00 04 36          .)..6
0.000000  750  00 8D 00 17 00 00 00 78 .......x

```

energee, 2017-04-09:
Tapped the CAN lines at the camera and I get mostly the same messages as the civic (was sniffing the can lines at the OBD port previously), except for 0x200 and 0x1fa: gas and brake. You can see them above this comment.

Trying to debug but it doesn't look like openpilot is reading or writing CAN messages, i get LKAS / ACC brake errors and 0mph when moving.

I added a fingerprint, and removing the fingerprint check, and tried running an always-on 12V line so i could restart the car with openpilot already running. Still same results.

Is there a way debug/log python code on the NEO and see where it is failing?

energee, 2017-04-18:
Noticed something different about the way the CR-V sends messaging vs the Civic. When the camera is unplugged, the only message that stops is 35E. When the radar is unplugged, 0xE4, 0xE8, 0x1DF, 0x1EF, 0x30C, 0x33D, and 0x39F stop sending.

So from what I understand, this is the reverse of the civic so the only way to solve this is to have a gateway in between the radar/ADAS and EPS, EBCM, and Gas control module.

geohot, 2017-04-19:
How did you unplug the radar?

geohot, 2017-04-19:
And ugh yea, the CR-V radar is connected to F-CAN. It's not on the Civic. That's annoying.

Easiest way to get openpilot working is to unplug the radar and install a separate external radar.

geohot, 2017-04-19:
And the 2016 CR-V isn't like this. 2016 CR-V should be easy.

TheMutley, 2017-04-19:
Or a CAN gateway and keep the existing radar as I suggested to energee. Did you had a chance to look at the CAN deciphering I sent you?

energee, 2017-04-19:
I'll try the gateway first, @geohot, civic radar or are there better options?

geohot, 2017-04-19:
Looked a bit, and I trust your analysis. What CAN program is that btw? I want to effectively rewrite that in javascript.

The Delphi ESR2.5 is nicer, you can find them on eBay for a couple hundred bucks, and the full DBC file is available online. We have some code to work with that radar we can include in 0.3 if you end up getting it.

The problem with a gateway is I still don't see that radar sending normal radar msgs. Does it send like positions of cars and stuff, or does it do all that computation internally?

energee, 2017-04-19:
I think it does send radar messages over the second CAN line, which goes directly to the camera harness but messages look different than the ILX/nidec Radar:

[image]

Also i don't see the delphi on ebay, but will look around.

geohot, 2017-04-19:
It's the Jaguar Land Rover radar iirc.

energee, 2017-04-20:
All XXXX-9G768-AD == Delphi ESR 2.5?

[BH32-9G768-AD](http://www.ebay.com/itm/JAGUAR-LAND-RANGE-ROVER-FRONT-ACC-CRUISE-CONTROL-RADAR-SENSOR-BH32-9G768-AD-/182507200331?hash=item2a7e46df4b:g:kU8AAOSwfY9XQflK&vxp=mtr) [BE9T-9G768-AC](http://www.ebay.com/itm/11-12-13-14-15-MKS-MKT-TAURUS-Cruise-Control-Module-BE9T-9G768-AC-MKT-ADAPTIVE-/182530841050?hash=item2a7faf99da:g:WhkAAOSwGy5Y8Ws2&vxp=mtr)

Do you happen to have pinout?

TheMutley, 2017-04-20:
George, I replied to you from my previous e-mail, radar and tracking information is there but it's mapped at CAN 4 on the log I sent you.

energee, 2017-04-22:
Can the radar operate through a plastic bumper / grill?

geohot, 2017-04-24:
Yes

energee, 2017-05-03:
Not seeing any CAN messaging on the delphi when "powered" up, could be a bad unit. I might try swapping in a tesla radar since its practically the same model and it would fit in the Honda mount and the connector.

Is there a Tesla radar DBC yet?

energee, 2017-05-03:
*Edit: Ran this by @TheMutley and didn't think about the fact that other OEM radars require PCM messaging like can speed, steering angle, etc.

Also found out i can't sniff a CAN line without an existing network or shorting it with a 120ohm resistor so will test more with the delphi.

energee, 2017-05-04:
btw @geohot, the radar in the 17 CR-V and Civic Hatchback is the Bosch MRR.

LuckyParker, 2017-05-04:
Do you think that once you get the crv working the hatchback might work
with some slight tweaking?

energee, 2017-05-05:
yes

energee, 2017-05-18:
@geohot can you upstream the delphi code, i have one.

energee, 2017-06-09:
I've made some good progress on this. Getting around the radar issue is tough and totally not shippable. Although I do have a Delphi radar, I may end up using a tesla radar because it drops right into the current Honda mount and i shouldn't have to worry about fascia issues. It has 32 tracks compared to the delphi's 64, but still more than the nidec. @jeankalud gave me some good direction for deciphering the tesla can messaging, but what I don't know is what messages like speed, etc will need to be sent to it. More to come.

vaxheadroom, 2017-06-17:
I new to this, but just wanted to point out that, based on the specs,  the Bosch MRR radar (on the '17 CR-V)  appears very similar to the Bosch Motorsport CAS-M system, and that the DBC file is available under the "Documents" folder at  http://www.bosch-motorsport.de/content/downloads/Products/18180530315.html#

energee, 2017-06-19:
@vaxheadroom This is a nice find, but the dbc doesn't match the honda firmware.

leo333, 2017-06-29:
Radar connect to bord,   There is no need  honda firmware for support.

shaitand, 2017-09-20:
So any updates?

energee, 2017-09-20:
Lateral Control works great, needs a radar for longitudinal control

haimiko, 2017-09-21:
No radar with panda?

energee, 2017-10-05:
Progress: https://www.youtube.com/watch?v=eMk4LNwiXro

haimiko, 2017-10-05:
Using Neo or Panda?  Just got my panda in the mail today, hoping to start working on mine.

shaitand, 2017-10-05:
So just to confirm. Using Neo or Panda? Using re-wired connector? Using Delphi ESR 2.5 Radar? Is this just using current release or do we need certain patches, etc?

If using external radar, any radar ending in this model (9G768-AD) that we can find will be legit?  What about the -AE and -AB? This adds quite a bit to the price, a couple hundred was mentioned but $300-$500 seems more common. If the -AE works you can buy them all day long https://www.aliexpress.com/promotion/promotion_adaptive-cruise-promotion.html

For the -AD it looks like 3 or 4 of us might get a better deal off E-Bay.

energee, 2017-10-05:
This is using the stock radar, a NEO and a panda, I might be able to eliminate the panda but you would definitely lose automatic lights/highbeams unless comma adds detection for them.

haimiko, 2017-10-05:
Is the NEO really necessary?  Can it not be done with Openpilot and Panda
only?

energee, 2017-10-06:
yes, actually, just one panda will work when this is fully complete, requires more code for message forwarding

haimiko, 2017-10-10:
What connector did you use for the Neo to the CRV?  I read that the
connector isn't like the civic.  Could you please point me to the wiring
diagram for the connector?

energee, 2017-10-10:
Pinout is from techinfo.honda.com and is in comment 5. You have to make your own until someone comes up with a better solution :/

shaitand, 2017-10-10:
@Ted Slesinski

Thanks again for all the great work you've done. Crimping pins and making a
connector doesn't seem that insurmountable. What are the actual connectors
involved if someone wanted to do so?

haimiko, 2017-10-12:
Ted,  I've got my panda, OP3 flashed with OpenPilot, Python coding skills and a 2017 CRV with sensing.   This is still really new for me. How do I proceed to start integrating panda into Openpilot?  Thanks.

KUASWoodyLIN, 2017-11-01:
Is it possible to have the giraffe product support crv2017 in the future?

energee, 2017-11-01:
@csouers working on a prototype:

[image]

stealthcrv, 2017-11-12:
Hey guys new here, I have a 2017 CRV EX-L w/Honda Sensing. How much progress has been made so far? Will be willing to help out testing if it’s ready to

csouers, 2017-11-12:
@energee has panda doing all forwarding now. At time of writing, it's only been a day or so since we've had that. Works here on my 2017 Civic Hatchback as well. We had V1 of our custom giraffe made and it's being tested. We're working on V2 of the board that will solve some issues that we have in V1.

[7 more replies]

closed 2018-05-23 by energee