commaai/openpilot #71: Add 2017 Acura MDX with AcuraWatch

easel, 2017-03-07:
I'm opening this to capture interest and find out if anybody else has attempted it. I assume the system should be exceptionally similar to the already supported vehicles, and the low speed ACC should resolve the speed control issue limitation reported on the ILX. 

Will report back as I learn more, I'm new to the space.

geohot, 2017-03-08:
I believe this car is the same as the Accord with steering. Needs a k-line transceiver

easel, 2017-03-08:
Thanks @geohot, any pointers to where I could start verifying this stuff? Are there some discussions about the accord floating around somewhere?

energee, 2017-03-09:
You need something to read the messaging on the Canbus like usb2can, pican2, CAN-BUS Shield or similar... sniff the packets on the CAN to track down the message and message ID that is fired for a particular event.

You'll have to look into the k-line transceiver, i'm assuming that means the NEO would need more hardware to control steering if the steering isn't controlled over CAN.

zhaohuaqing1993, 2017-03-20:
@geohot  Hi，i run the ./view_generative_model.py transition --name transition and always see the string "loading weights...",what should i do?thanks

geohot, 2017-03-20:
If you show me a NEO running with the gas and brakes working, we can help you out with some K-Line hardware.

JasonGiedymin, 2018-07-20:
For reference, the 2014 through to today’s 2019 could have a lot of the same hardware. I’ve driven every model year and the running characteristics of ACC and LKAS feel very close if not the same. 

I’d be willing to also capture bus traffic over these model ranges.

Why will be needed?

closed 2018-12-18 by vanillagorillaa

JasonGiedymin, 2018-12-18:
Sorry for the laaaate reply. I’ll joinnslack today and try to see if I can help out.