commaai/openpilot #75: Name duplication with OpenPilot flight controller

mlyle, 2017-03-16:
Hey everyone.  Sorry for opening another duplicate issue.  I think what you're doing is really cool and I'm tempted to try it and perhaps even be involved.

At the same time, I think you should really consider changing the name.  The OpenPilot firmware and related projects have tens of thousands of users and the duplicate use of the name is likely to cause substantial confusion.  While I was never involved in OpenPilot itself, I run a sister-project and I anticipate a lot of user confusion both ways from how our project uses the term "OpenPilot" in its documentation.  There's also 6 years of useful documentation about the project which faces being obscured in Google results from duplicate use of the name.

Also, the OpenPilot term is trademarked and is still actively used in trade -- https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20170316091352&SearchText=openpilot -- the deisgns that the project made are commercially available and still in active use.  While the core OpenPilot firmware is no longer under active development, that does not mean the project is inactive.

I think it's highly unlikely that someone will make a trademark claim against you guys, but they could.  But beyond that, you should please do the right thing.  You have a high number of watchers/stars/forks-- these will follow the project through any name change.  OTOH, there's not much others can do to move this 6 years of history to a new name.

Hell, you can probably pick a name that better describes what you do, and avoids repeating the mistakes / implications of Tesla Autopillot.  :)

geohot, 2017-03-16:
This is a duplicate issue, though I appreciate the reasonableness of this request. We currently have no plans to change the name. Despite claims to the contrary it looks to me like openpilot development continues under librepilot. And I really doubt the name confusion will cause a problem, one is for cars and one is for drones. I also don't see a problem with the Tesla Autopilot naming.

closed 2017-03-16 by geohot

mlyle, 2017-03-16:
> Despite claims to the contrary it looks to me like openpilot development continues under librepilot, with the openpilot github repo not contributed to since 2015.

You're absolutely right-- firmware development has ceased in the openpilot development repository.    However, there are still authorized vendors selling **hardware** under the OpenPilot name, and that hardware is referred to under the OpenPilot name in LibrePilot, dRonin, and associated projects.

> And I really doubt the name collision will cause a problem, one is for cars and one is for drones

You're right that most of the use is for drones, but OpenPilot does have autonomous ground vehicle support...

> many projects share the same name.

These types of things have a storied history of causing confusion and problems... e.g I don't know if you remember it at the time, but the Mozilla Firebird vs. FirebirdSQL conflict--- web browsers and RDBMS are fairly distinct spaces https://en.wikipedia.org/wiki/Firebird_(database_server)#Mozilla_Firefox_name_conflict

This is the type of thing that is already causing active confusion in the user community.

https://www.google.com/search?q=openpilot+ekf&oq=openpilot+ekf&aqs=chrome.0.69i59j69i60l2j0j69i60j0.1267j0j9&sourceid=chrome&ie=UTF-8

Your ekf.py looks really cool, and I hate our (dRonin's) dependence on MatLab code.  I'm tempted to extend it for native code generation and pull it in.  However-- it creates problems of attribution.  When we credit ekf.py from "OpenPilot"-- is it the same thing as the other 99.99% of OpenPilot code in our codebase?  What if any code flows the other way?

mlyle, 2017-03-16:
@d13-kevinf / @MAVProxyUser -- If you're authorized to act on the behalf of OpenPilot, it's pretty easy to make a trademark claim under GitHub's processes-- see here: https://help.github.com/articles/github-trademark-policy/ -- and they are pretty responsive to issues of this nature.  Of course, litigation is always possible, too, with opportunities to act in US Federal Court, under Australian Jurisdiction, or under WIPO processes.

tracernz, 2017-03-17:
That's a very unfortunate stance @geohot. :disappointed: [I'm not really sure that "open" is even an appropriate name for this project?](https://github.com/commaai/openpilot/tree/master/selfdrive/visiond). That's something very much in contrast with the real OpenPilot.