commaai/openpilot #79: Comma Unavailable:  Brake Error

jleonar, 2017-03-23:
Car Model:
2016 Honda Civic Sensing EX-L

Openpilot 0.2.9
Neo v2

Behavior: 
1.  Car already started
2.  Neo unresponsive (happens to me about once or twice a week)
3.  Reboot the Neo with the car still running
4.  Neo reboots completely
5.  Try to engage neo and the neo will throw a Comma Unavailable: Brake Error

Workaround:
1.  Turn the car off then back on.

geohot, 2017-03-23:
Yea, sadly if the car loses connection with the NEO it needs to be turned off and on again. The unresponsiveness shouldn't be happening though. We've been working on infrastructure to help us detect problems like this, any idea what causes it?

jleonar, 2017-03-23:
I haven't found anything consistent when it becomes unresponsive.  I don't normally notice until I start the car and all the honda error notifications happen.  

The only thing I can say for certain, it has only happened when my car has been sitting over night (8+ hours).

yugandhar91, 2017-03-23:
Happened to me too. Mentioned it in #67.

For me, it happens mostly when the NEO approaches a stopped vehicle with speeds >40mph and still accelerates as it comes closer because it can't see the vehicle in front up until it gets too close, then it starts braking harshly and with fear of collision, I start breaking too. Then it says controls mismatch or something similar and from then onwards I get this error trying to engage openpilot.

johnnwvs, 2017-06-15:
I noticed that this Brake Error (exactly that case - capital B, capital E) happens to me, and it is somewhat repeatable. I have a feeling it has to do with the number of steer saturation errors, or possibly any errors that spam the car network.

Today on my ride home, I allowed the system to brake on it's own and steer on it's own as I headed to a red light. Another car pulled out in front of me abruptly and the system started sending messages to the dash at a high rate of speed. The errors came in a wave so rapidly that it almost sounded like a continuous beep. I could not stop the beeps even when pressing the brake, pressing cancel, turning off LKAS, etc etc etc. I could not "make it stop". Eventually it game me a "controls error", or something similar. Once I get that error, I can no longer engage the system. I will only receive the "Brake Error" until I turn the car off and back on, and reboot the Neo for good measure.

So, is there a way to slow down the number of messages that go to the car? Something like one alert per 3 seconds or so? That might prevent this issue.

closed 2019-06-02 by pd0wm