commaai/openpilot #86: Gas interceptor & stock assist ECU

vadbutSwerve, 2017-04-14:
Hi,

Thanks for making it open-source! The self-driving field needs more enthusiasts like you!
I'm looking into trying it out for my ILX 2017. Two questions:
- do you have a description on how to implement a gas pedal interceptor?
- I noticed that my car assist ECU is constantly sending out CAN bus messages (e.g. 0xE4) even when the assists are off. Does your software somehow overwrite the stock CAN comms or I need to somehow fail my car ECU so it stops sending the comms?

Thank you!

geohot, 2017-04-14:
We unplug our stock ADAS system to send our 0xe4 instead. And gas pedal will work without interceptor, just not below 25 mph.

vadbutSwerve, 2017-04-14:
Thanks for the reply.
Can you give me a clue on where the ADAS unit is located?

geohot, 2017-04-15:
It's next to the rear view mirror, assuming you have that package.

closed 2019-06-02 by pd0wm