commaai/openpilot #98: 2016 CR-V Touring

energee, 2017-05-18:
Working on this with @johnwvs Fingerprint of the car here:

`0.200119   39  00 00 39                ..9
0.199962   91  80 64 48 01 FF 00 00 01 .dH.....
0.199961  13C  02 23 00 47 00 40 04 0E .#.G.@..
0.199989  154  12 00 00 FE A6 A9 00 0B ........
0.199972  156  00 BD 00 00 07 1C       ......
0.199833  158  00 00 00 00 00 00 00 0A ........
0.199940  17C  00 00 00 00 00 00 00 04 ........
0.199934  18E  00 00 6B                ..k
0.199988  18F  00 00 00 00 06 28       .....(
0.200010  191  01 00 00 A9 A9 01 01 04 ........
0.199953  194  00 00 00 0A             ....
0.199960  1A4  00 66 00 00 00 00 00 0D .f......
0.200104  1A6  00 08 C0 52 52 80 00 2B ...RR..+
0.199917  1AA  7F FF 00 00 00 00 00 0F ........
0.199971  1B0  00 00 00 00 00 01 0B    .......
0.000000  1D0  00 00 00 00 00 00 00 0A ........
0.199978  1DA  FF 7F FF 00 1D          .....
0.199980  1DC  42 00 00 35             B..5
0.200092  1E7  19 80 66 04             ..f.
0.199857  1EA  00 00 00 00 00 00 00 0F ........
0.200033  1ED  01 FF 3A                ..:
0.199924  1FA  00 00 00 80 00 00 00 15 ........
0.199954  1FB  0D                      .
0.200072  21E  00 00 3D 47 60 04 3F    ..=G`.?
0.200168  221  35 00 80 30             5..0
0.199989  255  00 00 00 00 55 55 7A DC ....UUz.
0.200008  294  04 16 00 00 A4 06 40 33 ......@3
0.200008  295  00 00 00 35             ...5
0.200046  305  80 54 00 00 00 00 1E    .T.....
0.199988  309  00 00 00 00 00 00 00 EE ........
0.199880  30C  00 00 8C FF 10 40 40 3B .....@@;
0.199814  320  00 00 00 00 00 00 00 30 .......0
0.199972  324  62 59 00 00 00 00 0E B0 bY......
0.199957  328  00 00 00 00 00 00 40 61 ......@a
0.199877  33D  00 00 00 40 3E          ...@>
0.200148  372  00 2A                   .*
0.199893  374  47 60 00 0B 00 00 2C    G`....,
0.199983  378  0F D2 04 19 25 87 FF 60 ....%..`
0.200002  37B  00 00 00 00 00 00 00 21 .......!
0.199991  37C  F6 00 00 FA 00 00 01 30 .......0
0.198222  39B  00 10                   ..
0.200080  3A1  40 00 00 00 00 00 00 15 @.......
0.199942  3D7  B0 00 00 00 00 00 00 33 .......3
0.199980  3D9  00 80 07                ...
0.300145  400  00 00 02 00 20          .... 
0.300247  403  00 00 00 00 2F          ..../
0.298651  405  00 00 00 00 00 00 00 0F ........
0.299780  409  00 00 00 01 28          ....(
0.300144  40C  09 00 00 00 00 00 00 2D .......-
0.300148  40F  01 0E 15 01 07 04 00 22 ......."
0.299944  421  00 00 01 01 2D          ....-
0.299292  428  06 00 00 00 A4 06 00    .......
0.300348  454  36 F9 02 29 30 00 80 20 6..)0.. 
0.300162  465  00 00 00 00 00 00 01 26 .......&
0.500637  510  00 00 00 00 00 00 00 11 ........
0.505430  555  00 00 00 00 27          ....'
1.000053  590  00 00 00 04 06          .....
1.000064  640  00 00 00 06 08          .....
1.000037  641  15 00 00 00 00 00 00 07 ........`

energee, 2017-05-18:
The obvious signals i don't see are:

"car_gas", "cruise_buttons", "main_on", "cruise_setting", "steer_torque", "steer_torque_request", and "gas_command"

geohot, 2017-05-19:
Does he have a NEO? Connect it up to log can.

energee, 2017-05-19:
Not yet

johnwaynehacker, 2017-05-21:
I have a NEOS,  what portion of FCAN can I update to refine the id, byte and bit of the exact steering, braking, gas, etc..,  what portion in the C/Python I can modified for different vehicle? What open source software do you guy use to make dbc file? other than Vector tool? I have a neos but need to update for different vehicle.

energee, 2017-05-21:
Just run the NEO and upload your can data to drive explorer

hanpingliang, 2017-05-21:
Anyone working on porting 2016 CRV?

@energee @geohot , could you tell me how to create the dbcs file based on the sniffed information? How to read the sniffed information?

Thanks,
liangpi

energee, 2017-05-21:
Steering looks like 194. Weird thing is that the only messages that stop sending (at the OBDII port) when cam is unplugged are:

```
194
1FA
30C
33D
```

Was expecting at least a fifth message for gas command. Might need to sniff directly on the camera's line. In the 17 CR-V, the steering message isn't exposed to the OBDII port, must be a gateway in between.

geohot, 2017-05-22:
The set speed is on 30C, same as the ilx and civic. This controls gas.

So looks like the only change is E4 -> 194

geohot, 2017-05-22:
And we made the dbc files using a text editor :)

johnwaynehacker, 2017-05-22:
I mention modified the vehicle code to allow cruise control under 25mph

geohot, 2017-05-22:
It's more than that. The message content is different as well.

There's no easy way to modify for the higher torques and low speeds required for city driving. And whenever you make modifications, always make sure you are following the safety guidelines in https://github.com/commaai/openpilot/blob/master/SAFETY.md

johnwaynehacker, 2017-05-29:
in the hondacan.py  where does it distinguish for which dbc file.  
194 message alive and counter checksum change.
the lsb steering factor change.
where do I change that? in here?

how can I change the lsb, and where is the pointer to point to certain dbc file in this command line?
"
def create_steering_control(apply_steer, idx):
  """Creates a CAN message for the Honda DBC STEERING_CONTROL."""
  msg = struct.pack("!h", apply_steer) + ("\x80\x00" if apply_steer != 0 else "\x00\x00")
  return make_can_msg(0xe4, msg, idx, 0)
"

johnwaynehacker, 2017-05-29:
Oh, it look like the steering command factor is 1 for 0E4 is ilx?  can I change it to more precise lsb factor? and the range is off.

how does the following command relate to 0E4? Some value is just 0 or 1, how come range is 127?
BO_ 228 STEERING_CONTROL: 5 ADAS
 SG_ STEER_TORQUE : 7|16@0- (1,0) [-3840|3840] ""  EPS
 SG_ STEER_TORQUE_REQUEST : 23|1@0+ (1,0) [0|1] ""  EPS
 SG_ SET_ME_X00 : 22|7@0+ (1,0) [0|127] ""  EPS
 SG_ SET_ME_X00 : 31|8@0+ (1,0) [0|0] ""  EPS
 SG_ COUNTER : 37|2@0+ (1,0) [0|3] ""  EPS
 SG_ CHECKSUM : 35|4@0+ (1,0) [0|3] ""  EPS

energee, 2017-06-01:
Good news coming soon for this car!

johnwaynehacker, 2017-06-03:
@energee That's good news!
Are you working for comma.ai?
I can send you the dbc file update for CRV.
What fingerprint and control file should I modified to read new dbc for 2015-2016 crv?
I update gas pedal lsb, brake value, steering lsb request range on 194, update checksum count number.

johnwaynehacker, 2017-06-03:
Crv have cvt transmission, content same as Civic

energee, 2017-06-03:
We have a completed dbc, will upstream soon.

johnwaynehacker, 2017-06-04:
@energee that is great. 
I modified the accord for crv car controller, because the range is +-9,

what is this does? #add 2015-2016 CRV touring
    elif CS.crv2015:
      idx = frame % 2
      can_sends.append(hondacan.create_crv2015_steering_control(apply_steer, idx))  
      #end 2015-2016 CRV touring

I modified  carcontroller,carstate and hondacan & make new dbc file.
How can access the driveexplorer, the neo oneplus 3t one doesn't show up.
[filemodified.zip](https://github.com/commaai/openpilot/files/1050643/filemodified.zip)

johnwaynehacker, 2017-06-04:
sorry why idx = frame % 2? and other one %4?  if the lsb of 194 is 
BO_ 228 STEERING_CONTROL: 4 ADAS 
 SG_ STEER_TORQUE : 7|16@0- (0.000305,0) [-10|10] ""  EPS 
 SG_ STEER_TORQUE_REQUEST : 23|1@0+ (1,0) [0|1] ""  EPS  
 SG_ SET_ME_X00 : 22|1@0+ (1,0) [0|1] ""  EPS  
 SG_ SET_ME_X00 : 31|1@0+ (1,0) [0|1] ""  EPS 
 SG_ COUNTER : 29|2@0+ (1,0) [0|3] ""  EPS  
 SG_ CHECKSUM : 27|4@0+ (1,0) [0|3] ""  EPS   ?? what checksum is 0 to 3 not 0-10 or 0-15?, does another file account for the checksum?

And does Openpilot have any checksum error if I plug into the car?

energee, 2017-06-04:
`BO_ 399 STEER_STATUS: 7 EPS
 SG_ STEER_TORQUE_SENSOR : 7|16@0- (1,0) [-31000|31000] "tbd"  NEO
 SG_ STEER_TORQUE_MOTOR : 23|16@0- (1,0) [-31000|31000] "tbd"  NEO
 SG_ STEER_STATUS : 39|4@0+ (1,0) [0|15] ""  NEO
 SG_ STEER_CONTROL_ACTIVE : 35|1@0+ (1,0) [0|1] ""  NEO
 SG_ COUNTER : 53|2@0+ (1,0) [0|3] ""  NEO
 SG_ CHECKSUM : 51|4@0+ (1,0) [0|15] ""  NEO`

abdulrahman004, 2017-06-05:
@energee is your work for the 2015/16 CRV Touring only ?

energee, 2017-06-05:
The 15/16 touring is the only CR-V model in these years that has honda sensing.

johnwaynehacker, 2017-06-06:
@energee @geohot 
How do modified the openpilot file and upload to the NEO hardware myself? It seen like it install using openpilot.comma.ai .  How can reinstall new version.

I drive to open the oneplus 3t manually to access drive. But it seen it can't recognize as an android device. 

Sorry maybe this answer somewhere, but I haven't figure it out.

energee, 2017-06-06:
@johnwaynehacker ssh into the phone, code is in /data/openpilot

energee, 2017-06-06:
https://drive.google.com/file/d/0B_XTioiTDzikSWc5cjlLNTN1ZGM/view

energee, 2017-06-08:
#102

johnwaynehacker, 2017-06-12:
@energee 
 
 2015-2016 CRV and 2016 & 2017 Accord is the same system on 4cylinder vehicle.
Have this work on 2016 Accord 4 cylinder with LKAS? It has the same 194 message across the board for accord just transmission message change.
CRV transmission gear message is break out bit by bit.

@energee @geohot 
2. regarding ssh into the phone, i'm keep getting connection time out.
I bought the neo from neodriven, is there anyway to root the neos back to factory oneplus and reinstall neos v3, maybe my key not correct.

or is there to install custom openpilot.comma.ai with custom file change to test the code.

command line use:
ssh -p 8022 -i key/id_rsa 192.168.0.25
key file id_rsa  and id_rsa.pub

pjlao307, 2017-06-12:
You can run your own custom code by cloning your fork of openpilot.  That's what I've done.  Just make sure your custom code is in the /data/openpilot directory on the neo.  Granted, this doesn't help you if you can't even SSH into the phone.  Sometimes I can't connect to the neo via SSH either.  

First thing to check is if the IP address of the neo is correct.  It may have changed.  Can you see the list of DHCP clients on your  wifi?

johnnwvs, 2017-06-12:
You can also check the IP address on the Neo. I can't remember exactly
where, but it is somewhere under settings.
My Neo IP address lease just changed today.

John

LuckyParker, 2017-06-12:
@johnwaynehacker I did not buy a neodriven but you should be able to just reflash it with the most current neo is following the same steps. I did this to mine recently.

johnwaynehacker, 2017-06-12:
The IP address I can check on the neos. It show the IP address. I wondering if is the key issue or my wifi.

energee, 2017-06-12:
@johnwaynehacker that is interesting that the accord has the 0x194 message, might not be a huge effort to port that year. Do you own one of these?

johnwaynehacker, 2017-06-13:
I own it, but transfer to step sister, but I can borrow and test.  I still haven't problem getting neos code to open. Is here any to access one plus 3t without ssh. I'm using a Mac to ssh to it. Unless port forwarding issue on router. I'm haven't check port forwarding on router setting yet.

johnwaynehacker, 2017-06-13:
latest try by adding port 8022 and 22 ssh net to neos device ip port forwarding.
get this error and then stuck:
The authenticity of host '[192.168.0.19]:8022 ([192.168.0.19]:8022)' can't be established.
ECDSA key fingerprint is SHA256:5peozHRu3R3E8OzoS4dwhHBlWJX3v8EbjBq90zpCCEE.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '[192.168.0.19]:8022' (ECDSA) to the list of known hosts.
root@localhost:~$ ls
root@localhost:~$

energee, 2017-06-13:
@johnwaynehacker why dont you join the comma [slack team](commaai.herokuapp.com), someone there will be able to help you with ssh, its a bit OT for this thread.

johnnwvs, 2017-06-21:
After the latest code changes all the safety features are back in and the Honda CR-V is working well with these latest changes. I uploaded a video of the latest test:
https://youtu.be/6IW7Nejsr3A

With these latest changes, we believe the port is complete.

johnwaynehacker, 2017-06-26:
I try noticed for the CRV in high wind system, the steering is off most of the time.
Which part of lat g setting can change to account for it? It drifts toward the center line too much, and if the highway left side has uneven road layout, if tend to over correction and need to take control.﻿

Also,
 when it brake by itself.  I get brake error, and I have turn on and off the car to able to set it again.

energee, 2017-06-26:
Try adjusting the `ret.steerKp, ret.steerKi = 6.0, 0.5` values on line 130 of fingerprints.py. It sounds like these variables could be adjusted lower if it is overcorrecting too much.

johnnwvs, 2017-06-26:
It did that for me as well until I had several hours of driving time down. And every time I changed my mount for the Neo it had to be re-trained with several hours of driving. Not just several hours, but also uploading all the data from the previous drives. I am not sure how long it takes after you upload a drive to have the new calibration take effect. I just know after a couple of days it would steer better.

johnnwvs, 2017-06-26:
John,

Most of what you describe sounds like a lack of sufficient driving to train the system. How long have you been driving for and uploading your drives back to comma.ai? Are you changing the way you mount the Neo in any way, or does it tend to stay in the same place?

The brake error seems to happen to many cars, not just the CR-V. You can find people talking about this in other treads.

johnwaynehacker, 2017-06-27:
I been leave the mount at one location. Couples hours first day, and couples hours second days. Mainly I only drive it on the weekend. This holiday I will  drive more. For the calibration? Does it need to be driving using the neos it self or it need me to drive it for several hours and then upload the data overnight, then drive using the neos again?

energee, 2017-06-27:
The latter, but calibration doesn't need several hours, only about 30-45 minutes are required using the NEO (openpilot does not have to be engaged) then upload that data to the cloud.

closed 2017-06-28 by energee