commaai/openpilot #99: Clarity surrounding the Honda Civic steering firmware limitation

mrmayfield, 2017-05-22:
In the readme it says

> Limitations: Due to limitations in steering firmware, steering is disabled below 18 mph

Can I get some clarity on what exactly that means? Does the openpilot system disengage completely? Does it hamper usability to the point where openpilot can't really work when you're going slower than 18 mph? 

Just a bit of understanding as to the impact of what that limitation is would be great, thanks & great project!

geohot, 2017-05-22:
It just means it can't command steering torque. The system will remain engaged and beep if it wants to steer when it can't. (it may be more like 12 and not 18 actually)

mrmayfield, 2017-05-22:
It can't command any steering torque below that limit hey? Does it have a big impact on driving? And maybe we can find with more certainty the speed below which the vehicle is unable to independently steer because of the steering firmware, if I look at this as someone who is interested in this project the Honda Civic 2016 seems to be the natural choice because of the Acura's speed limitation, so if we can be more precise with the Civic limitations we can reduce the barriers to entry people like myself may have as we begin to scope out the project.

pjlao307, 2017-05-24:
I've visually confirmed that openpilot steering will work (at least on my 2016 civic touring) down to 12mph.  Once the speedometer hits 12mph steering will stop working until it gets above 12mph again.

mrmayfield, 2017-05-24:
Great, thanks a lot for the clear explanation. How do you find that the steering limitation affects usability, how big is the impact during your typical usage?

pjlao307, 2017-05-25:
It hasn't been a problem for me.  When it's going that slow I'm typically sitting in stop and go traffic which means it's going straight ahead most of the time.  If i'm sitting at a curve I may have to adjust the steering abit because it's slowly drifting to one side of the lane or the other but I never have to touch the gas because the adaptive cruise is always active (unless I manually cancel openpilot).

closed 2018-12-18 by vanillagorillaa