commaai/openpilot #100: Open source IMX298 camera driver

ingenieroariel, 2017-06-01:
Would it be possible for commaai to publish the IMX298 camera driver for linux as an open source kernel extension independently from the visiond binary?

I own a Jetson TX2 and 3 x $6 usd Oneplus 3 rear camera, I think it should be possible to replace the camera it has with that one if I can figure out the correct wiring and have two of them taking stereo video at high fps and the third one taking high resolution video at 30fps.

geohot, 2017-06-01:
Hmm, don't you work for our competition now?

I'm not even sure what you mean by camera driver. Most of our camera stack is qualcomm specific, and already open source in the Android kernel. The user space stuff in visiond is highly qualcomm specific as well. It won't really help you with an nVidia port.

For prototyping, why use such a difficult camera? Just get some Point Grey.

closed 2017-06-01 by geohot

ingenieroariel, 2017-06-01:
In order to load specific cameras on the jetson tx2 or tk1 I need to enable a .ko kernel module, that's what I am calling the driver for lack of a better word.

Right now I am using a close sourced one from e-con systems or the one nvidia ships with, that one was quite expensive and I'd like to move to cheaper (and better) ones.

I'd like to see myself as a comma ai coopetitor, I'm still working on creating the DBC file and publishing detailed info about each can command as I understand them: https://github.com/autti/abraham/issues/63 and updating issue #1 here.

geohot, 2017-06-01:
We didn't write any kernel code for our camera driver, it's already open source.

https://android.googlesource.com/kernel/msm/+/8a49608cf010892935740e26fe575dde56e5224e/drivers/media/video/msm

geohot, 2017-06-01:
(our full kernel is also open source, and we don't load any modules)

https://github.com/commaneos/android_kernel_oneplus_msm8996

ingenieroariel, 2017-06-01:
This is great news to me, I'll look for something that looks like the `imx074.c` in the official android folder on the commaneos fork.

Thanks a lot!

espes, 2017-06-01:
https://github.com/commaneos/android_kernel_oneplus_msm8996/tree/master/drivers/media/platform/msm/camera_v2