commaai/openpilot #105: Is it not too difficult to add support for a Hyundai Sonata 2011 Sedan?

OrlandoTroncoso, 2017-06-15:
I would like to do my own research since I own that car and add support for it in openpilot. I am a firmware engineer so I have software skills but not car hardware skills. Would this be too difficult or a big job?

energee, 2017-06-15:
Does the sonata have lane keep assist?

OrlandoTroncoso, 2017-06-16:
No, it does not but I hear that has electric power steering. I am guessing that means it has a motor that I can control and I can send it steering angles/commands. So I do not need anything else from the car to do lane keep assist. Software that somebody else has written will do the job for me.
I do not know if it brakes by wire or if the gas throttle is by wire. I cannot find this information. Do you happen to know how I can find out?

energee, 2017-06-16:
The easiest way to port is with cars that have lkas and acc but the community is working on ways to port recent cars without these. Your car would probably fit into this group.

OrlandoTroncoso, 2017-06-16:
My main goal is to learn, I am taking Machine Learning and deep learning classes. After I finish them I am doing the self-driving car nano degree at Udacity. So I don't even want the car to have lkas or acc. The only thing I need the car to have is a way to control the steering, the breaks, and the gas. The rest I am going to build software and learn from other people's software how to do. Now sadly I am not a car guy so I do not even know what features my own car has. How can I find out what can be controlled electrically in my car? I am trying to find out if I can control steering, gas and brakes. Any help would be appreciated. Of course I plan to contribute what I learn to the source code here.

TheMutley, 2017-06-17:
Could you post a CAN log from your car or from any other Hyundai ?

OrlandoTroncoso, 2017-06-17:
I do not have the equipment for that yet but I am going to order it and do just that. I am ready to buy anything that I need to get this done and learn more. I'm very technical (firmware engineer) but know zero about cars and mechanics. Also probably this is not the right place for me to be posting these questions. Should I go to the forums instead or is it ok to have it here?

energee, 2017-06-17:
Commaai.herokuapp.com

energee, 2017-07-14:
Comma is selling the panda now. panda.comma.ai

closed 2019-06-02 by pd0wm