commaai/openpilot #107: neo PCBs manufacturing order

alan707, 2017-06-24:
Hey All,
I'm thinking about making a small batch of neo PCBs sans the OnePlus.  I am getting quotes to make a small run of 50 units. 

I'd rather just get the PCB assembled and ready to go rather than soldering myself. I can do print the case with my 3D printer. 

Would anyone want an already assembled neo PCBA?

telescopeuser, 2017-06-24:
Sounds cool.
Actually I am interested in buying a fully built one, including all:  OnePlus 3 smartphone and an STM32F205-based CAN interface board, in a 3d-printed housing with active cooling.

Is your offer the fully built one?

alan707, 2017-06-24:
you can get that from neodriven for $1000. I'm just thinking selling the CAN interface board.  Case should be easy to make and you can get a used OnePlus off ebay. The PCB seems the most painful to me that is why I'm thinking if we could make a bulk order would be nice.

telescopeuser, 2017-06-24:
Agree with you the most painful part. My getting-old eyes and hands start to complain for this work.
Thank you the info on sourcing places.

alan707, 2017-06-24:
👍

ingenieroariel, 2017-06-24:
I'd like to get two!

alan707, 2017-06-24:
So far we have 5 spoken for. Nice!

LuckyParker, 2017-06-25:
You should probably post this on the forums instead opening an issue with the repo. Use the link below.

https://community.comma.ai/

dickhfchan, 2017-06-26:
Hi, I am new to Neo and got some questions. Is openpilot supported by Android? Is 3T the only Android phone supported?

LuckyParker, 2017-06-26:
That question is located in the guide here:

https://github.com/commaai/neo

If you have any questions please feel free to start a thread at the forum site here:

https://community.comma.ai/

gowthamrao, 2017-06-27:
I would be interested in one or two

alan707, 2017-06-27:
Cool, so far 7! We need another 43 to run a batch somewhat economically.  I'm thinking $59 per PCB and we can get these hammered out.

pjlao307, 2017-06-27:
Another forum user named latimer also builds these boards for the community, you may want to get in touch with him to see if he'd be interested in this batch purchase.

On a side note, George has stated the NEO board is Gen 1 and the Panda is Gen 2 of the CAN interface for the NEO so keep that in mind.

alan707, 2017-06-27:
cool, thanks for the heads up. I will reach out. Also, are you referring to the Giant Panda? It seems to use the Panda which has a closed source PCB design. Or at least I haven't found the schematics for it.

pjlao307, 2017-06-27:
No, I'm not referring to the Giant Panda.  There is a hardware guide in the Panda repo here:

https://github.com/commaai/panda/blob/master/docs/guide.pdf

Is that what you're looking for?

alan707, 2017-06-27:
Not what I am looking for.  There is no schematic or PCB layout.  I think it is intentionally closed source?

edsammy, 2017-06-28:
Panda software is open but board is closed. We open sourced the neo since it was a simple design with a few (relatively) easy to solder components. The board is also simple enough that you could make it on something like the Othermill or even etch it yourself.

Pandas are much harder to manufacturer (wink) with all of the QFN packages and 0402 passive components so open sourcing it doesn't make much sense. Just sit tight for an update soon.

Can we either move this discussion to [slack](https://comma.slack.com/) or the [neo repo](https://github.com/commaai/neo) since it is not openpilot related?

alan707, 2017-06-28:
yes absolutely!  didn't know where it would be the best place.  Which do you recommend?

edsammy, 2017-06-28:
Slack seems best. Its getting quite active now. We passed our 10k message limit!

dickhfchan, 2017-07-03:
@edsammy , is the panda hardware in mass production? I am interested in purchase it.

closed 2018-12-18 by vanillagorillaa