commaai/openpilot #111: adjustable car steering or steering not read straight

johnwaynehacker, 2017-07-03:
How do adjust actually car steering alignment reading? the blue one is off from straight in the road, it causes the steering to over correct too much and veer off road.

What value can adjust the steering reading from car? or does it read straight from dbc? then I just need to add offset?
[image]

what value does the machine learning to read and adjust steering to be small smooth instead of side to side adjustment steering when it drive straight, how can I improve it to be smoother?

mwkrieger, 2017-12-27:
I'm seeing the same thing. I wish there was some sort of 'trim' function to tweak things slightly.

mfd00gie, 2018-03-10:
Same problem here 16 civic, alignment is a bit out though

CRV-Driver, 2018-04-07:
Same here, 2015 CRV. So far OpenPilot is a no-go until it can stay on the road.
I'm not seeing much response from Comma.ai on this.  From July 2017 and as of now it's April 2018. You would think that keeping the car on the road would be a big deal for them???

closed 2019-03-13 by legonigel