commaai/openpilot #112: NEO Paired False in OpenPilot Setting

WaliedA, 2017-07-04:
We have been getting NEO Paired = False in OpenPilot Setting, we also figured out the comma points are not redeemed in the account while using NEO. Only Chffr is redeemed.

We are using NEO in a non supported car with no connection to CANH ot CANL only power and ignition

pjlao307, 2017-07-04:
You won't get comma points unless the video includes CAN messages on the neo.

geohot, 2017-07-04:
Hmm, you should, but only 1 per minute.

WaliedA, 2017-07-04:
So I have connected it to Range Rover Sport, If I connect the CANh and CANl would it count the comma point, keeping in mind car has no ACC/LKAS

energee, 2017-07-14:
It should give you 2 points when wired to CAN on an unsupported vehicle but I've seen some mixed behavior when only connected to one CAN. Seems to inconsistently show up in drive explorer _most_ but not _all_ of the time after being uploaded. Try connecting to CAN and post the results.

closed 2019-06-02 by pd0wm