commaai/openpilot #135: Ask experts here for some suggestions

hbzhang, 2017-09-04:
I am working for a University in U.S. For teaching and undergraduate research purposes, we are interested in making a self-driving car. 

After digging through, I found CommaAI is a good starting point.  Due to the limits budget, we want to minimize the cost. So can someone here give suggestions for a good starting point?

We are thinking of Buying Nissan Leaf 2018, which itself has quite advanced automation capability and use Open-pilot, NEO or Panda, for a starting point. Is this a good starting point?  or even doable? We aim to use a half year to a year to make it self-drivable.  Please advise.

Thank you!!

energee, 2017-09-05:
If the leaf can control pcm, ebcm, and eps, it is a good candidate for openpilot, generally any cars with lane keep assist and adaptive cruise control can do this. The next next step is to get access to the tech info on the car and find out how the radar, camera, and/or adas modules send control at https://www.nissan-techinfo.com/home.aspx. Cars that control via CANbus are easier since comma has put together some excellent tools for can analysis. The panda can also control over LIN if necessary.

Join the slack team at comma.slack.com, #carports for more specific questions.

hbzhang, 2017-09-05:
Thank you, Ted, greatly appreciates your information! Just joined slack, talk to you then.

geohot, 2017-10-01:
We will also sell a 75k Prius devkit that includes full gas, brake, and steering control, as well as DBC files for a big list of sensors including radar.

closed 2019-06-02 by pd0wm