commaai/openpilot #144: ImportError: No module named opendbc

Capri2014, 2017-10-23:
This is what I've got by running `/.run_docker_test.sh`, seems like the PYTHONPATH was not set right?
```
Successfully built b4ab4237c798
Successfully tagged tmppilot:latest
Traceback (most recent call last):
  File "./test_longitudinal.py", line 14, in <module>
    from selfdrive.test.plant.maneuver import Maneuver
  File "/tmp/openpilot/selfdrive/test/plant/maneuver.py", line 4, in <module>
    from plant import Plant
  File "/tmp/openpilot/selfdrive/test/plant/plant.py", line 8, in <module>
    from opendbc import DBC_PATH
ImportError: No module named opendbc
```

energee, 2017-10-23:
Run `git submodule update --init --recursive`, I hear they moving to git subtrees on next release.

Capri2014, 2017-10-23:
@energee  Thanks! It works!

closed 2017-10-23 by Capri2014