commaai/openpilot #148: 2017 Elantra (Hyundai)

TK211X, 2017-11-07:
The 2017 Hyundai Elantra Ultimate has LKAS and ACC. I think for me the holly grail would be just to circumvent the 12 second annunciation to that requires you to keep your hands on the steering wheel (counter input) for the system to remain active.

In this vehicle:

LKAS activates above 40mph when both lane markings are seen by the system. There are three options in the cluster for LKAS. The most sensitive one can very decently keep the car in the middle of the lane and through gentle-moderate curves.

ACC can be enabled above 7Mph and disables below 7Mph - It does not have stop and go.

I have no codeing or programming experience and I would like to know if this is a feasible attempt with Panda since I assume there isn’t any support for Hyundai’s.

I’m thinking of getting it. Would anyone on here be able to help me if I post all the logs ect. and collaborate with someone who is knowledgeable. Or will Panda completely not be able to interact with my car?

Looking for some guidance - Thanks.

TK211X, 2017-12-03:
Bought it even though community here is dead. Will try anyway.

energee, 2017-12-03:
you'll need to work on a dbc first, join http://slack.comma.ai

vanillagorillaa, 2018-12-18:
Upstreamed, closing

closed 2018-12-18 by vanillagorillaa