commaai/openpilot #153: 2017 Honda Accord HondaSense EX-L 4cylinder

devindied, 2017-11-14:
Looking to make the neo work for my Accord. Any suggestions? I'm sure I need to sniff the CAN bus. Any suggestions on what to do? Total noob to the scene. I'm willing to offer up my car to further the project. Has anyone made any progress with the Accord? What hardware do I need to add for the steering(if any)?

shaunbaker04, 2017-11-16:
Honda Models Currently Featuring Honda Sensing

2016 Honda Accord
2016 Honda Civic
2016 Honda CR-V
2016 Honda Pilot

shaunbaker04, 2017-11-16:
https://automobiles.honda.com/sensing

devindied, 2017-11-16:
I'm aware of what the car does from the factory. I'm trying to get the neo to control the vehicle.

energee, 2017-11-17:
The accord can be ported without extra hardware (aside from an EON, panda, and giraffe), there is a lot already completed on the [DBC](https://github.com/energee/opendbc/blob/master/honda_accord_touring_2016_can.dbc). If you are technical, you can finish the port; there is not a lot left to do.

devindied, 2017-11-22:
I'm more than willing to give this a try. However, I can not do it without your assistance. Do I need the neo, panda, and giraffe? Or just the neo?

What is the first step? 

If you guys want this to be ported to the Accord, let us know. And if you're technically inclined, help us! I live in Houston.

zongbinlin, 2018-02-01:
saw the twitter and got a new 18 accord, let me know what i need to try the open pilot

billy2kool, 2018-02-21:
Anyone working on 2016 Honda Accord Touring.  I got the Panda, The EON and Giraffe plug everything in and nothing happened.  Need help please oh I already install open Pilot 4.2 on the EON

energee, 2018-02-22:
2018 accord is a different generation than 2016 and 2017. Its possible comma might be writing an official port for the 2018, but afaik, this is not confirmed. The 2016 and 2017 is similar to the other Hondas, however steering is controlled over a 9600 baud serial line. You would want to understand what is being sent over the lines by sniffing the EPStoLKAS and LKAStoEPS lines and replicating the messaging in hondacan.py.

closed 2018-12-18 by vanillagorillaa