commaai/openpilot #164: How do I make 2017 Prius Prime to be a self driving vehicle?

fiftycentsjj, 2017-12-22:
Hi Sir: 
          Greeting to you.
          Eon+Panda+Giraffe+Chrrf. Can the vehicle be a self driving vehicle with that all? Do I need to install openpilot in to EON? How? Your kindly reply would be appreciated. Thank you so much.

geohot, 2017-12-22:
EON is designed to be a dashcam. It is not designed to drive a car.

closed 2017-12-22 by geohot

fiftycentsjj, 2017-12-22:
Many thanks for your quick reply. Any possibilities to make 2017 Prius Prime self driving as what you show Honda self driving video on the website? Can I install openpilot in to EON with Giraffe, Panda and Chrrf for making vehicle self driving? I am really interested in your self driving solution. Thank you very much.

anhhna, 2017-12-22:
@fiftycentsjj as I know your car can if it have the Adaptive Cruise Control and Advanced Lane Keeping System likes Honda Civic