commaai/openpilot #173: Needed bounties;

Mr-Rubicon, 2018-01-16:
Toyota; 

86, Tacoma

Subaru; 

BRZ, WRX

Capri2014, 2018-01-30:
do you consider adding mazda?

bertramt, 2018-01-31:
If your vehicle isn't on the list of the top 10 cars sold in the last couple years, I wouldn't expect comma to place a bounty or put much effort into a port. The biggest problem is older cars lack the required drive by wire abilities that comma supports.  

At best you could rally a community bounty in the forums or slack channel.   See the [candidate vehicles](https://community.comma.ai/wiki/index.php/Candidate_Vehicles) page on the wiki for some guidance on what vehicles might be capable of self driving.  Visit #old_cars on the comma slack for discussions on older cars.

bpanicker13, 2018-03-27:
Need a bounty for Highlander Hybrid

andrewfree, 2018-04-18:
Is there any technical reason the new Subarus w/ eyesight would not work? Seems like a good one to add.

geohot, 2018-06-24:
Bounty program is cancelled due to effort required to maintain and test safety of ports.

closed 2018-06-24 by geohot