commaai/openpilot #181: how to support Honda FIT?

9crk, 2018-01-29:
as a young man ,you know, FIT is so popular.
here is my Questions about supporting FIT:

1.can the autopilot work without radar?
2.can the OBD control the steeling wheel on FIT? if not, what if I design a motor to control the steeling wheel ?
3.can the autopilot software be installed on most of the mobile phones?what's the hardware requriement?

bugsy924, 2018-01-29:
https://community.comma.ai/wiki/

vanillagorillaa, 2018-02-05:
1. Not currently. ACC with Radar is needed until Vision Only support is added.
2. Does the car have EPS? On most Honda's steering only occurs from the LKAS camera and not OBD2
3. The OnePlus 3/t is the only supported hardware for OpenPilot

jfrux, 2018-06-26:
#266

closed 2018-12-18 by vanillagorillaa