commaai/openpilot #182: openpilot EON screensize right side keyboard mapped off screen

kmacpher67, 2018-01-30:
https://plus.google.com/+kenmacpherson/posts/QbUeAeNuxKT
upgrade EON to openpilot 0.4.0.2 and keyboard screen gets pushed aside

From the Comma.ai main screen, click "Log in" and google sign in page appears.
click focus into email/phone number field 
keyboard right of the letter "K" is forced off the screen, looks portional to left hand nav.

bugsy924, 2018-01-30:
Known issue on initial setup
Reboot EON to load small keyboard

kmacpher67, 2018-01-30:
I was just coming here to post. 
REBOOT DOES NOT DO IT. 
SHUTDOWN. LOL. 
I did a reboot before posting this too. 
Sorry. 
Funny, that the shutdown basically does a reboot. I wonder if it needs to be done twice. 
Anyrate, log in works! Now on to getting my Linux laptop enabled.

closed 2018-01-30 by kmacpher67