commaai/openpilot #187: 2017 Prius auto high beam missing

hailite, 2018-02-06:
After enabled OP, all other functionalities work  fine, good job! However, auto high beam functionality is missed.

mwkrieger, 2018-02-26:
I saw this on my Honda as well. Turns out, the sensor is in the rearview mirror, so of your EON is blocking that sensor, or you've removed the rearview mirror - your auto high beams won't work.

hailite, 2018-02-27:
@mwkrieger I turned on rearview mirror, but auto high beam still didn't work.

csouers, 2018-02-27:
Incorrect. The camera provides auto high beam functionality on Honda, likely on Toyota as well. The sensor on the back of the mirror is for auto-dimming rear view mirror only. If you power off the camera through giraffe, you lose auto-high beam functionality. Retaining it would require message forwarding and keeping the camera alive.

billy2kool, 2018-02-28:
I have a 2016 Honda Accord touring and I think your right.  But is there anyone working on touring.

chandroos, 2018-03-22:
@csouers, is there a way you can retain the power to the camera? Any panda settings?

bitshop, 2018-03-25:
Is the camera just sending something to say it sees light or doesn't? We can likely simulate AHB with a bit of modeling...

arne182, 2018-03-26:
I would like to work on this after I have finished with the blindspot sensors. I assume it will be similar. The same problem exists on the RAV4 Hybrid.

mtourangeau, 2018-05-28:
Is there a way I can help? I'm not famillialr with linux that's why I didn't fix any problem yet. If you can help me setting up the debugging environment on Windows, I'll help to fix this problem. I already install Visual Studio Code but I don't know how to debug python scripts.

arne182, 2018-05-29:
Capturing auto high beam events would help. Set the stock adas on and drive
at night over 40km/hr with the AHB on. We need to find out which can
message is being sent. You could search through the messages with cabana or
share the drives where the events occurs.

mtourangeau, 2018-05-29:
Perfect. I'll do that.

mtourangeau, 2018-05-30:
How do I record my drive if I use dip switch 1101? OpenPilot will not record my drive with this setting. Should I use 0101 for procy stock adas w/ panda?

andy867, 2018-05-30:
The panda will still be recording all the can messages. Are you using comma power by chance?

mtourangeau, 2018-05-30:
Can I use Chffr on iPhone connected to my panda to capture code?

mtourangeau, 2018-05-30:
I think I have what you need. Here it is:
https://community.comma.ai/cabana/?route=96515cca9b5ce486|2018-05-29--22-30-25

mtourangeau, 2018-05-30:
I can't find the CAN message with the link I provided... :(

mtourangeau, 2018-05-30:
I'm using eon with panda and Giraffe. But for the recording, I connected my panda to the standard odbii port, install Chiffr app on iPhone, connect the Chffr app to the wifi panda's and record my drive. Did it works?

arne182, 2018-05-30:
You need to use the "copy share link" button. If you could also give the
time at which the ahb come on and off that would be good

mtourangeau, 2018-05-30:
https://community.comma.ai/cabana/?route=96515cca9b5ce486%7C2018-05-29--22-30-25&max=3&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2F96515cca9b5ce486%2F6d435809c4dee8a0598bbae5e7c2c48a_2018-05-29--22-30-25

at 8 seconds the high beam come ON
a 14 seconds the high beam come OFF

arne182, 2018-10-25:
It should be working now on my branch. AHB

hailite, 2018-10-25:
@arne182 Are you able to send a PR? Thanks.

arne182, 2018-10-25:
The fix is so easy and thanks to wocsor and otherchase I could. The fix has a caveat.
 It requires a 120ohm resistor on pins 12 and 13 of your panda. Then enable message forwarding in panda firmware and remove static values for camera.

bitshop, 2018-10-25:
Resistor because we need to terminate the CANBUS? If so we should make a 
plug in connector with M/F plugs and resistor in between so no soldering is 
required.

Steve Radich

arne182, 2018-10-26:
Yes. Termination is needed. Sounds nice. For now I just have it jammed into my giraffe and the Panda forced in. It does not even look that bad.

arne182, 2018-10-27:
Pull request done. https://github.com/commaai/openpilot/pull/411
Test and comment.

vanillagorillaa, 2018-12-18:
added this functionality in 0.5.6. Closing

closed 2018-12-18 by vanillagorillaa

arne182, 2018-12-18:
Requires 120ohm resistor but all the information is in the pull request. What is up with the massive cleanup vanillagorillaa?