commaai/openpilot #204: Couldn't Run PC simulation outside docker

hrubaiyat, 2018-03-02:
Hi, I'm working on the PC version of Openpilot. I successfully ran the simulation, but it runs inside a docker daemon. I wanted to use my own lane detection algorithm as vision model but while running inside docker it couldn't recognize some of the libraries that are installed in my PC. So, I wanted to run it without the docker support. I resolved the dependencies and path issues. The code runs but shows 'Assertion 'dbc' failed' message from parser.cc file. I tried different things inside the Makefile of selfdrive/can directory, even copied the whole opendbc folder to the tests/plant directory, but couldn't get it work. What am I missing here? Also is there any other way so that the simulator can recognize the executables/libraries from directories outside the docker?
Thanks

ed2k, 2018-03-04:
you need to make your own visiond run inside docker. that is much easier than porting simulation outside.

hrubaiyat, 2018-03-04:
I tried. But I need opencv libraries for the visiond and openpilot cannot find these libraries while running inside the docker. I tried to setup opencv libraries inside docker like the other libraries listed in requirements text file, but the installation gets aborted because of large size. It's also not feasible to setup opencv every time I run openpilot.

hrubaiyat, 2018-03-08:
Finally solved the issue. Need to modify the OPEN_DBC variable inside the makefile in CAN directory. That resolved the issue. Now it's working outside docker and can access the local libraries and executables. Thanks.

closed 2018-03-08 by hrubaiyat

fiftycentsjj, 2018-03-13:
@hrubaiyat, I could run docker test in Ubuntu already. Meanwhile, what did you modify the variable in the makefile of open_dbc? I am trying to make the outside docker? Thank you.

hrubaiyat, 2018-03-14:
@fiftycentsjj, I needed to change the makefile for OP versio 0.3.8.2, but didn't face this issue in latest version (0.4.2). So, in this version you don't need to change anything in makefile. To run OP out of docker you have to follow the the following procedure - 
1. First install all the libraries in your PC.
     - apt-get update && apt-get install -y build-essential clang vim screen wget bzip2 git libglib2.0-0 python-pip capnproto libcapnp-dev libzmq5-dev libffi-dev libusb-1.0-0
     - pip2 install numpy==1.11.2 scipy==0.18.1 matplotlib
     - pip2 install -r requirements_openpilot.txt # this will install the libraries mentioned in the requirements text file
2. Create a separate .sh file and include the following lines in it,
     export PYTHONPATH="$PWD":$PYTHONPATH
     rm selfdrive/can/libdbc.so  # if you don't need to modify the .cc files or the makefile inside can folder, then you can ignore this line
     /bin/sh -c 'cd selfdrive/test/tests/plant && OPTEST=1 ./test_longitudinal.py'
3. Then execute this new .sh file instead of run_dockers_test.sh file. You may need to use sudo to execute this new sh file.

This is the basic procedure. You may see some module or path issues whole running the python file, but that are solvable. Hope this will help you. Thanks

fiftycentsjj, 2018-03-14:
@hrubaiyat , Many thanks for your directions. Let me try first. I will let you know what the result is after running. Thank you so much.

fiftycentsjj, 2018-03-14:
May I have your slack account for sending screenshot to you? Thanks.