commaai/openpilot #211: How to gather training data for the compute_gb neural network?

sibojia, 2018-03-10:
In the car interface module I notice that the acceleration is converted to the actual gas/brake command by a small neural network. In the code comments `see debug/dump_accel_from_fiber.py to see how those parameters were generated` however this file doesn't exist.

I want to tweak the NN parameter for my car but not sure how the data should be collected, taking into consideration that the command/sensor has latency and the speed of the car can not be easily kept constant. Could you share the `dump_accel_from_fiber.py` script or give me some guidance on how to do this by myself? Thanks very much!

rbiasini, 2018-03-12:
Hi @sibojia , that file is obsolete and we are moving away from using NN to predict gas and brake, given accel desired and vehicle speed. I recommend collecting some data and come up with more explicit functions. To mitigate latency in data, you can make some test by commanding constant brake or gas. The fact that speed is changing shouldn't be an issue.

sibojia, 2018-03-16:
@rbiasini Thanks! I'll try it.

closed 2018-03-16 by sibojia

HaoZhouGT, 2021-02-15:
@rbiasini Hi may I ask a related question here? Except for current speed, why don't we include current acceleration when searching for the gas/brake for the desired accel? I'm assuming the current acceleration would also have a great impact unless a constant speed is assumed. Can you share some comments on this? It's clever to command a constant brake or gas and collect the real acceleration values. Just curious why we don't include aEgo as an input [aEgo, vEgo, desired accel] ---> gas/brake?