commaai/openpilot #212: Getting steerSaturation warning in PC simulation

hrubaiyat, 2018-03-12:
I'm running the PC simulation of the latest version (0.4.2), but getting the 'steerSaturation' warning/alert message. Then I found a line inside controlsd.py file (controlsd_thread function) that states 'angle_offset = 1.5' which makes the output_steer greater than the steer_MAX from the beginning. Previously I was working on version 0.3.8.2 and there it was 'angle_offset = 0.0'. I replaced the current code with this one (i.e. replaced 1.5 with 0), now there is no steerSat alerts. Can anybody explain why the bias is set 1.5 instead of 0.0. Is there any other way I can avoid the alert without replacing the bias value.
Thanks

rbiasini, 2018-03-12:
@hrubaiyat, your explanation for steering saturation alert is correct. Steer offset by default is set to 1.5deg. This is because in average (on multiple cars) that's what the steering offset converges to (our vision model is slightly biased to the right, so you need a steering offset to the left to be centered in the lane...). Setting the steering offset to 1.5deg by default improves (in average) the system performance right out of the box, as steering angle offset calibration takes a while to converge.
There are many ways to remove the alert during simulation: the simplest is probably to hardcode the simulated steering feedback to 1.5deg so the controller does not winds up.

fiftycentsjj, 2018-03-13:
@hrubaiyat , I can not set the OP simulator in either Ubuntu or Windows  successfully. Could you guide me how to set up the environment for running it? Your kindly help would be appreciated. Thank you very much.

hrubaiyat, 2018-03-13:
I used the PC simulation in Ubuntu 16.04. I followed the procedure written in the README file. That's all. I faced some issues that I forgot and didn't record, but that was some missing library issues. Can you share your procedure and the errors you are facing? Thanks

fiftycentsjj, 2018-03-13:
@hrubaiyat, in Ubuntu 16.04 I got the message as followings. Could you let me know where you got the readme.txt for running your simulator? Thank you very much.
./plant.py
  File "./plant.py", line 211
    print "%6.2f m  %6.2f m/s  %6.2f m/s2   %.2f ang   gas: %.2f  brake: %.2f  steer: %5.2f     lead_rel: %6.2f m  %6.2f m/s" % (distance, speed, acceleration, self.angle_steer, gas, brake, steer_torque, d_rel, v_rel)
                                                                                                                            ^
SyntaxError: invalid syntax

Meanwhile, I ran the ./run_docker_tests.sh under the folder of openpilot. followings what I got.
-bash: ./run_docker_tests.sh: /bin/bash: bad interpreter: No such file or directory

hrubaiyat, 2018-03-13:
I followed the readme file in the github repo. There's nothing much for PC simulation other than './run_docker_tests.sh' command. You don't need to run any of the python files directly. ./run_docker_tests.sh command should do. For the bad interpreter error, I'm not quite sure. Did you download/clone the codes directly from git to Ubuntu? Or copied it from other OS i.e. MAC or Windows?

fiftycentsjj, 2018-03-13:
@hrubaiyat , Yes, I cloned whole the folder of openpile-devel from git under Ubuntu. Do you mean that run_docker_tests.sh could be ran in Ubuntu?

hrubaiyat, 2018-03-13:
run_docker_tests.sh is the script you need to run the PC simulator. It will create the environment and execute the python files.

fiftycentsjj, 2018-03-13:
@hrubaiyat, May I have you steps for running the simulators?

hrubaiyat, 2018-03-13:
I just do the './run_docker_tests.sh' from the openpilot directory.

fiftycentsjj, 2018-03-13:
@hrubaiyat, I got it, Let me try again. Thank you so much for your help and information.

hrubaiyat, 2018-03-13:
Best of luck. Sorry, for not being able to more helpful. Let me know the progress. Thanks

fiftycentsjj, 2018-03-13:
@hrubaiyat , Ok, I will. Do you have slack account for connecting you directly?

closed 2019-06-02 by pd0wm